libuvc/libuvc #12: libuvc on Windows

sureshjoshi, 2015-02-06:
Has anyone successfully streamed webcam images from this library on Windows?

I've done the necessary porting to make it compile, and installed WinUSB drivers (as per libusb), but claiming the interface when I want to stream returns an ERROR_NOT_SUPPORTED and if I hack to pick an interface that claims correctly, I get an ERROR_PIPE.

This is a Logitech C270 camera, running the example.cpp and test.cpp, and works perfectly on Linux using the same computer.

Thanks!
-SJ

Timmmm, 2015-06-08:
There are probably two issues:

1. When you plug in a UVC webcam on Windows it installs Windows' native UVC driver. libusb requires a WinUSB-compatible driver interface on Windows so that doesn't work.

2. You should be able to remove the drivers Windows automatically installs (via the Device Manager) and install a WinUSB-compatible one - the easiest way is using [Zadig](http://zadig.akeo.ie/) - libusb supports all three driver backends that Zadig will install (libusbK, WinUSB and libusb-win32). These drivers all expose the WinUSB API.

If you're wondering the difference, WinUSB is the official microsoft "generic" USB driver, libusbK is an open source implementation of WinUSB, and I'm not sure about libusb-win32. [This page](http://libusbk.sourceforge.net/UsbK3/usbk_comparisons.html) has some comparisons.

Anyway, the second problem is that libusb doesn't support isochronous transfers on Windows (which I assume UVC uses). Although the libusbk backend *does* support isochronous transfers, the libusb library ignores that and treats all WinUSB-compatible backends equally. However I asked about this on the mailing list and there are two possible solutions.

[Solution one](http://marc.info/?l=libusb-devel&m=143311324202510&w=2) is to use [a modified version of libusb](https://github.com/JoshBlake/libusbx/commits/winiso) that directly supports libusbk and therefore allows isochronous endpoints.

[Solution two](http://marc.info/?l=libusb-devel&m=143315876131628&w=2) is to use *yet another* USB driver - [UsbDk](http://cgit.freedesktop.org/spice/win32/usbdk) which supports isochronous transfers (and is also easier to install than the others). There is another modified version of libusb that supports this backend.

For my own project I am going to try solution two as I think it will lead to a nicer user experience. By the way I did notice some things about "filter drivers" which may mean you don't need to uninstall Windows' native UVC drivers but that is a total guess.

There is yet *another* generic USB driver

Timmmm, 2015-06-08:
By the way, I'm going to have a go at this. Would you mind posting your modifications?

mkassner, 2015-06-08:
@Timmmm This is very relevant for us as well. 

Will you develop in a fork? We would like to contribute and test. 

@MichaelBarz What do you think?

sureshjoshi, 2015-06-08:
@Timmmm Thanks for all the info. Unfortunately, I moved my development over to native cross-platform drivers for the project I was working on, and for the life of me, I can't find my libuvc mods... Sorry.

However, I know Steven Lovegrove (https://github.com/stevenlovegrove/libuvc) started a Windows version too.

Timmmm, 2015-06-08:
Ok I will have a look at his fork. I wish github let you give forks descriptions. Sometimes it's really hard to work out what the difference between them all is! @mkassner if I get it to work I'll post here. Not sure about code as it is for work and they're a bit paranoid about IP. Hopefully it won't require many code changes anyway.

flecheria, 2015-06-09:
I describe what I've done until here and also I need some further information.
It's a lot of stuff for a post but I hope it could be helpful for people like me that try libuvc for the very first time.
1 - uninstall old drive and I made a fresh installation of *WinUSB* with Zadig (**Is it suffice or I have to install all? libusbK, WinUSB and libusb-win32?**) following @Timmmm suggestions
2 - download *libusb* from [libusb.info](http://libusb.info/)
3 - put it on this path on my machine *C:\Users\Paolo\Documents\software\libusb-1.0.19-rc1*
4 - cloning *libuvc* (obviously ;-) )
5 - change the path on **CMakeLists.txt**
```
find_library(LIBUSB_LIBRARY_NAMES usb-1.0
  PATHS C:/Users/Paolo/Documents/software/libusb-1.0.19-rc1/MinGW64/dll)

find_path(LIBUSB_INCLUDE_DIR libusb-1.0/libusb.h
  PATHS C:/Users/Paolo/Documents/software/libusb-1.0.19-rc1/include)
```
**Questions:
-- @mkassner in my case i need libuvc to build pyuvc, is it the dll the right choice?
-- I choose the MinGW64 because I will build *libuvc* using tdm-gcc 64 bit
-- there is a way to write the find path instructions for every OS?
-- can I use the find_path to indicate also the path to jpeg?**
6 - open CMAKE, set the path for code and binaries (as you can see in the picture), configure and generate the code using MinGW Makefiles as generator. reading the results, It seems that now libuvc is compiled at 64 bit, but I'm not sure.
[image]

**Does CMake an automatic choise of the 64bit compiler refering to the MinGW64/dll path?**
**What is CMAKE_GNUtoMS? Do have I to check the options?**
7 - goes to libuvc/build and type from the console
```
mingw32-make
mingw32-make install
```
All seems to work. I will run some tests in next few days.

[image]

**last stupid question: mingw32-make is reffering to 32bit or 64bit or it has nothing to do with this because all is already setup in CMAKE when i generate?**

Timmmm, 2015-06-09:
@flecheria WinUSB won't work because it doesn't support isochronous endpoints (required for UVC) until Windows 8 (and libusb doesn't take advantage of that support anyway). libusbK *does* support isochronous endpoints but again libusb doesn't take advantage of that support. However there are old patches for libusb that change it to add support for libusbK's isochronous API. This is "solution one" that I linked before.

So you (or I) need to port those old libusb patches to the current version (shouldn't be too hard), then compile that version of libusb, install the libusbK driver using Zadig, and then hopefully it should work. The downside of this method is that no other apps will be able to use the webcam because you'll have to uninstall the system UVC driver.

The other option - UsbDk which I linked before as "solution two", might work too and might not require removing the system UVC driver. However I just had a bad experience with it so I am going to avoid it for now!

flecheria, 2015-06-14:
@Timmmm I rebuild libusb with the patches as you suggest on solution 1, uninstall WinUSB and install libusbK. It seems works.
Do you have a snippet of code for testing? because I have no idea on how to test isochronous transfer...
Thanks

Timmmm, 2015-06-15:
There is an isochronous benchmark example here:

https://github.com/libusb/libusb/blob/master/examples/sam3u_benchmark.c

One of my uses for isochronous transfers with libusb is for an application
using the SAM3X processor (which happens to be the one used by the Arduino
Due). I ported the Arduino Due to mBed and have an isochronous transfer
working (well, as far as I can tell; haven't tested libusb yet), but the
code isn't really in a shareable state yet.

Probably the easiest way to get something that sends isochronous transfers
is to buy an mBed device with USB support. The cheapest is the FRDM-KL25Z
<https://developer.mbed.org/platforms/KL25Z/> which I don't own but I just
looked at the code and it seems to support isochronous transfers.

Then you can modify one of the mBed USB examples to just write isochronous
data. It's not too difficult.

Or you could wait a week or two until I have time to give it a go. Or just
try it with libuvc!

Cheers,

Tim

flecheria, 2015-06-16:
Oh Thanks
I'll give it a try, I'm not sure to be properly skilled for thi stuff, but I'll try this week end.

Timmmm, 2015-06-17:
I had a go at this. It got fairly far - submitting the iso transfers
worked, but the transfer status in the callback was LIBUSB_TRANSFER_ERROR.
I stepped through and found that in windows_handle_events() io_result was
getting set to 87, which means "parameter is incorrect". I haven't been
able to work out why. I'm going to ask on the libusb mailing list.

Timmmm, 2015-07-08:
I asked on the mailing list but no reply... I may brave UsbDk again...

Timmmm, 2015-07-19:
Ok, I downloaded libusbK and tried using it directly rather than through libusb and it worked! Therefore it seems to me like the libusb patches to add support for libusbK isochronous transfers are incorrect in some way. I haven't investigated what they do differently to the libusbK example code.

MichaelBarz, 2015-07-29:
@Timmmm, sounds great! I recently tried to build libuvc on Windows too, but without success. Could you share your steps on how you replaced libusb by libusbK? And what is you system setup?
I use a VM running Windows 8.1 Pro x64, Visual Studio 2008 and libusbK 3 dev kit.

Best, Michael

maurosanjo, 2015-08-05:
Same here. I managed to compile libuvc and libusb. I can enumerate devices, connect, and when I try to start streaming I get UVC_ERROR_NOT_SUPPORTED, that came from inside libusb (LIBUSB_ERROR_NOT_SUPPORTED). 

Tried with WinUSB and libusbK drivers using Zadig.

I'm checking libusbK, does any of you guys managed to port libuvc to use libusbK?

Timmmm, 2015-08-06:
Yeah maurosanjo - that is the expected response (see my previous comments).

I've tested the libusbK api directly (not with libuvc) and it worked, and I've tested libuvc using libusb with a libusbK backend that supports isochronous transfers but it didn't work.

Probably easier and better to fix the libusbK backend for libusb than to port libuvc to libusbK.

Alternatively if you don't care about windows 7 you could upgrade to windows 10 and add in the isochronous support in libusb for the WinUSB backend.

MichaelBarz, 2015-10-13:
Hi. I tried to get this running again.

I used libusbK (Zadig), Windows Release of libusb-1.0.

There are a few things that I could solve or circumvent:
* Not all UVC-classified devices can be opened with libusb on my system, thus I modified `uvc_get_device_list` in order to only allow those that opened successfully.
* The DEBUG prints do not work for Windows. Dropping the line and file information solves this.
* Some further functions that print status information cause a crash on Windows, e.g. `uvc_print_diag`. Can be commented.
* It is possible to open a UVC device, but not to start a stream. `uvc_claim_if` fails because of `libusb_detach_kernel_driver` that itself is not available on Windows: http://libusb.sourceforge.net/api-1.0/group__dev.html#ga21bd343325f558987ca57e4e281a6d47

AFAIK there are no kernel drivers for Windows. I'll try to leave the corresponding function calls for detaching and attaching again and report if I had success.

timlukins, 2016-02-09:
Just tried Windows build of 0.0.5 of libuvc based on Windows 1.0.20 libusb binaries release from official site here: http://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.20/libusb-1.0.20.7z/download 

Everything still exactly as reported by @MichaelBarz above. I used Zadig and tried the WinUSB and libusbK driver. I also had to disable the debugging macros (but then able to get debug build). Everything with querying the camera works - but streaming breaks on `libusb_detach_kernel_driver(devh->usb_devh, idx)` in `uvc_claim_if`.

This is a real shame and a bit of showstopper...

Looking through commit log it doesn't seem like there has been any updates on the master branch that might address this.

Might it actually be a problem with the release build of libusb then? My next step was going to go down the rabbit hole of creating a windows debug build of libusb to find where `detach_kernel_driver` goes...

timlukins, 2016-02-09:
Digging into the libusb code for 1.0.20 says it all really...

**libusb/core.c**

**/libusb/os/windows_usb.c**
```
static int windows_detach_kernel_driver(struct libusb_device_handle *dev_handle, int iface)
{
        return LIBUSB_ERROR_NOT_SUPPORTED;
```

mkassner, 2016-02-09:
@timlukins I dont think this call makes sense on windows. It should be omitted since windows does not know the concept of a 'kernel driver' AFAIK . 

I dont have the knowledge or time to get this to work on windows and my suspicion is that the real problem comes down to getting this to work: https://github.com/libusb/libusb/issues/46

We really want to see libuvc working on windows and plan to but a (substantial) bounty on this. Interested?

Timmmm, 2016-02-10:
I could get libuvc to work on Windows 10 or later (well actually Windows 8). Did you mean you plan to *put* a substantial bounty on it? If so I'm interested (assuming you don't require Windows 7 support).

mkassner, 2016-02-11:
@Timmmm if that is the case we should talk! Please contact me: moritz [at] pupil-labs dot com

kshatrea, 2016-03-03:
Has anyone been able to solve this?

willpatera, 2016-07-07:
There's a bounty on this now :point_right: https://www.bountysource.com/issues/35223500-add-windows-support

isikmustafa, 2017-04-18:
Is there any updates on this?

mkassner, 2017-04-20:
We made this fork work for windows: https://github.com/pupil-labs/libuvc You will need to use libusbK drivers.

reda-alaoui, 2017-04-28:
Hello @mkassner, 
can you create PR to merge your modifications in the main repository?

mkassner, 2017-04-28:
@reda-alaoui  I can do that but we have diverged form the main repo on two other issues so I m not sure how clean that will be. I ll make a PR and you can cherry pic what you like out.

asdlei99, 2017-05-03:
@Timmmm @mkassner  uvc_start_streaming  ret UVC_ERROR_NOT_SUPPORTED , i have a try above, but still error

asdlei99, 2017-05-03:
env:  windows10  vs2013 x64  debug

willpatera, 2017-05-03:
@asdlei99 - did you follow the instructions in https://github.com/pupil-labs/pyuvc/blob/master/WINDOWS_DEVELOPER.md ?

JulianLiao5, 2017-05-12:
@asdlei99 Do you have further investigation, do you succeeded? I still fail by following the instructions on https://github.com/pupil-labs/libuvc. Please give me an hand, thanks!

JulianLiao5, 2017-05-12:
@mkassner @Timmmm I follow the instructions on  https://github.com/pupil-labs/libuvc, but still fail uvc_start_streaming ret UVC_ERROR_NOT_SUPPORTED(-12), can you guys give me a hand ? Thanks in advance!

JulianLiao5, 2017-05-12:
@mkassner @Timmmm @asdlei99

JulianLiao5, 2017-05-12:
@Timmmm @mkassner @asdlei99 

by the way, I use libusbK driver + pupil-labs/libusb + pupil-labs/libuvc, it still fail.

JulianLiao5, 2017-05-12:
@willpatera I follow the instructions in https://github.com/pupil-labs/pyuvc/blob/master/WINDOWS_DEVELOPER.md, it talks about python pyuvc, it has no business with the issue we discuss, and it can solve our problem.

ftaiolivista, 2017-10-19:
Hi I'm on Windows 10. I have compiled Pupil LibUSB fork, then Pupil LibUVC and then created a test Visual C++ app.
I have used Zadig to apply LibusbK to webcam (I see the webcam under libusbk devices).
I'm able to open device and query info but when a try to start streaming I keep getting Not Supported (-12).

What I'm missing? :(

[test.txt](https://github.com/ktossell/libuvc/files/1399157/test.txt)

XinyuKhan, 2019-05-31:
I get this error too,bro. Do you have further investigation?

cn-newit, 2020-03-06:
Is there any new progress?

JulianLiao5, 2020-03-06:
sorry, guys, I forget all the details due to it's an about 3years-old issue and no document.

ftaiolivista, 2020-03-06:
I was able to make it working on windows a couple of years ago using this fork https://github.com/ftaiolivista/libuvc

cn-newit, 2020-03-06:
OK, I'll give it a try!
Thanks.

ftaiolivista, 2020-03-06:
Found some notes that can help you.

Use LibUSB fork from pupil labs, follow this instructions:

https://github.com/pupil-labs/libusb/blob/master/INSTALL_WIN.txt

Use my LibUVC fork (  https://github.com/ftaiolivista/libuvc ) following this instructions:

https://github.com/ftaiolivista/libuvc/blob/master/INSTALL_WINDOWS.md

You may need turbo jpeg:

https://github.com/libjpeg-turbo/libjpeg-turbo.git

waddlesplash, 2020-03-07:
I asked pupil-labs about upstreaming their changes here (https://github.com/pupil-labs/libuvc/issues/34) and got no response, so I may try to just merge those manually, unless someone else would like to submit a PR.

PeterPeterPan, 2020-09-30:
Anyone use the usbdk driver on Windows?
I use this fork: https://github.com/pupil-labs/libuvc
step1: operate UsbDk_1.0.22_x64.msi
step2: modify the func uvc_init() in init.c (Reference from the libusb wiki)
```
libusb_init(&ctx->usb_ctx);
libusb_set_option(ctx->usb_ctx, LIBUSB_OPTION_USE_USBDK);
```
step3: run my code

I got the log:

> "uvc: device has a status interrupt endpoint, but unable to read from it"

any idea? thanks!

KrCool, 2020-10-12:
Excuse me, I see you are talking about uvc and libusb.

I need to use libusb in Windows to send data to the uvc camera (at the same time I need to use opencv to play the camera screen), I am now facing a problem:
1. If the driver is the uvc driver that comes with windows, then I cannot use the libusb library based on C language to open the USB device correctly
2. If I use libwin32, libusb, or libusbk to install the driver, I can open it, but I cannot use opencv to play the screen normally.
It's so hard, I feel my hair is saying goodbye to me

ftaiolivista, 2020-10-12:
Hi. In an old project I managed to use libusbk + libuvc + opencv on windows in a C++. I'm able to liveview webcam and send USB commands to the same device.
But you cannot use directly the webcam with Opencv VideoCapture API, you have to use LIBUVC as middle layer to caputer images and then you can use opencv to elaborate them.

KrCool, 2020-10-12:
> 你好。在一个旧项目中， 我设法在一个窗口使用 libusbk _ libuvc _ opencv C++。我能够实时查看网络摄像头，并发送 USB 命令到同一设备。
> 但是，您不能直接使用网络摄像头与 Opencv 视频捕获 API， 你必须使用 LIBUVC 作为中间层来封顶图像， 然后您可以使用 opencv 来详细说明它们。

Hi, thank you for your prompt, I have seen the relevant information of libuvc, but have not decided to use this method. I think I should try this method

KrCool, 2020-10-12:
> Hi. In an old project I managed to use libusbk + libuvc + opencv on windows in a C++. I'm able to liveview webcam and send USB commands to the same device.
> But you cannot use directly the webcam [...]

Oh，In addition, there is a situation where I can use a simple uvcplayer to play the screen and use the data sending function of Bus Hound at the same time. They work well together.

palydawn, 2020-12-16:
> Oh，In addition, there is a situation where I can use a simple uvcplayer to play the screen and use the data sending function of Bus Hound at the same time. They work well together.

Hi, I encountered the same problem as you, did you find way to send command to uvc camera without changing uvc driver?

ftaiolivista, 2020-12-16:
> Hi, I encountered the same problem as you, did you find way to send command to uvc camera without changing uvc driver?
no, sorry

KrCool, 2020-12-16:
> Hi, I encountered the same problem as you, did you find way to send command to uvc camera without changing uvc driver?

I haven't found a way to send commands directly, but you can use the method defined in DirectShow, Dshow.h to control the camera(it's send the relevant command automatically,but you can't define your own), and it won't affect your ability to use other processes to play pictures

THYAGARAJAN786, 2021-05-12:
> Is there any new progress?

i know its late,had the same issue UVC_ERROR_NOT_SUPPORTED, the fix was using zadig to replace both the composite and normal driver for the camera, under zadig list all devices -> ensure the option "Ignore hubs or composite parents" is unticked, a warning message would come, choose yes and replace the composite driver, Mine worked on libuvc(master), libusb(master) and windows10 64-bit release build, Hopefully helps someone else

mcuee, 2021-08-06:
> i know its late, had the same issue UVC_ERROR_NOT_SUPPORTED, the fix was using zadig to replace both the composite and normal driver with (WinUsb driver) for the camera, under zadig list all devices -> ensure the option "Ignore hubs or composite parents" is unticked, a warning message would come, choose yes and replace the composite driver, Mine worked on libuvc(master), libusb(master) and windows10 64-bit release build, Hopefully helps someone else

Yes, this is most likely the known issue of libusb https://github.com/libusb/libusb/issues/85 .
Using WinUSB to replace the USB composite parent driver is the current recommended workaround.

Or you can try this libusb pull request.
https://github.com/libusb/libusb/pull/965

mcuee, 2021-08-06:
vcpkg libuvc formula is supposed to work but somehow it does not work for me.
https://github.com/microsoft/vcpkg/tree/master/ports/libuvc

mcuee, 2021-08-06:
https://github.com/pupil-labs/libuvc is still easier to use for Windows users. You do not need to use the libusb fork there, rather you can use the official libusb.

mcuee, 2022-05-08:
> Latest build of https://github.com/pupil-labs/libuvc git head with libusb-1.0.26 (dynamic link) and pthreads-w32-2-9-1-release, using VS2019 (both 32bit and 64bit). 
[pupil_labs_libuvc.zip](https://github.com/libuvc/libuvc/files/8646865/pupil_labs_libuvc.zip)

yomei-o, 2022-12-30:
I succeed to work a UVC webcam on Windows.
I think this issue is libusb bug.

[https://github.com/yomei-o/libusb/commit/6d30e874a8748be7bfbaffb2f1a5821d39bb83a3](https://github.com/yomei-o/libusb/commit/6d30e874a8748be7bfbaffb2f1a5821d39bb83a3)

mcuee, 2022-12-30:
> I succeed to work a UVC webcam on Windows. I think this issue is libusb bug. And I tested two UVC devices. (isochronous transfer device and bulk transfer device)
> 
> I used libusb v1.0.26 and libuvc v0.0.6
> 
> [yomei-o/libusb@6d30e87](https://github.com/yomei-o/libusb/commit/6d30e874a8748be7bfbaffb2f1a5821d39bb83a3)

If you think there is a bug in libusb, please create a PR against libusb git HEAD here.
https://github.com/libusb/libusb/pulls

You should fork libusb project and the modify from there. Nobody will be able to understand what you are doing in your repo.
https://github.com/yomei-o/libusb/commits/main

mcuee, 2022-12-30:
Instructions to build libuvc under Winodws and MSYS2. I have not tried it myself.
https://github.com/libuvc/libuvc/issues/245

[6 more replies]