libuvc/libuvc #14: License?

teknotus, 2015-02-21:
I cannot find a License file. What are the terms of use, and contribution?

ktossell, 2015-02-21:
Wow, I'm really surprised that I didn't include a license file. Libuvc is distributed under the BSD 3-clause license. Contributions should be licensed the same way; contributors don't need to assign copyright or anything.

I'll add a license file or a readme entry, but in the meantime, you can find license headers in most of the src/*.c files if you need an in-distribution licensing statement.

suertreus, 2015-07-09:
Thanks for creating this and making it available under such a generous license =).

Please do add a LICENSE file listing or linking to the text of the license.  Doing so is very helpful to packagers who want to redistribute the code without worry and to businesses who want to build commercial or free products.

ktossell, 2015-07-10:
Does this work? https://github.com/ktossell/libuvc/commit/68612e2ccbe5889d21aa50c597c3aae2823fb5d4

suertreus, 2015-07-10:
Perfect, thanks again!

closed 2019-07-06 by waddlesplash