libuvc/libuvc #16: uvc_stop_streaming() hangs on Mac OS X

sadaszewski, 2015-04-17:
I originally detected it in my code but even in the attached example.c application, sometimes it exits but sometimes it just hangs in uvc_stop_streaming(). The backtraces from all the threads are included below. Interestingly an application which uses libuvc internally - oaCapture doesn't seem to have issues starting/stopping the stream (they're using libuvc 0.0.5 and so am I). Could you please take a look at this situation. It's a great library but this weird behavior is calling for some ugly workarounds (like running capture in separate process) which I think everybody would rather avoid. The SIGSTOP is coming from my Ctrl+C in LLDB.

(lldb) thread backtrace all
* thread #1: tid = 0x3a8e7, 0x00007fff8d3d3716 libsystem_kernel.dylib`__psynch_cvwait + 10, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  * frame #0: 0x00007fff8d3d3716 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #1: 0x00007fff8fca4c3b libsystem_pthread.dylib`_pthread_cond_wait + 727
    frame #2: 0x000000010000cd8b libuvc.dylib`uvc_stream_stop + 171
    frame #3: 0x000000010000c897 libuvc.dylib`uvc_stream_close + 23
    frame #4: 0x000000010000ccc9 libuvc.dylib`uvc_stop_streaming + 25
    frame #5: 0x0000000100000d2c a.out`main + 492
    frame #6: 0x00007fff8da195fd libdyld.dylib`start + 1

  thread #2: tid = 0x3a901, 0x00007fff8d3cfa1a libsystem_kernel.dylib`mach_msg_trap + 10, name = 'org.libusb.device-hotplug'
    frame #0: 0x00007fff8d3cfa1a libsystem_kernel.dylib`mach_msg_trap + 10
    frame #1: 0x00007fff8d3ced18 libsystem_kernel.dylib`mach_msg + 64
    frame #2: 0x00007fff93e6ef15 CoreFoundation`__CFRunLoopServiceMachPort + 181
    frame #3: 0x00007fff93e6e539 CoreFoundation`__CFRunLoopRun + 1161
    frame #4: 0x00007fff93e6de75 CoreFoundation`CFRunLoopRunSpecific + 309
    frame #5: 0x00007fff93f23811 CoreFoundation`CFRunLoopRun + 97
    frame #6: 0x00000001000551b9 libusb-1.0.0.dylib`___lldb_unnamed_function66$$libusb-1.0.0.dylib + 679
    frame #7: 0x00007fff8fca2899 libsystem_pthread.dylib`_pthread_body + 138
    frame #8: 0x00007fff8fca272a libsystem_pthread.dylib`_pthread_start + 137
    frame #9: 0x00007fff8fca6fc9 libsystem_pthread.dylib`thread_start + 13

  thread #3: tid = 0x3a902, 0x00007fff8d3d4662 libsystem_kernel.dylib`kevent64 + 10, queue = 'com.apple.libdispatch-manager'
    frame #0: 0x00007fff8d3d4662 libsystem_kernel.dylib`kevent64 + 10
    frame #1: 0x00007fff9164e421 libdispatch.dylib`_dispatch_mgr_invoke + 239
    frame #2: 0x00007fff9164e136 libdispatch.dylib`_dispatch_mgr_thread + 52

  thread #4: tid = 0x3a903, 0x00007fff8d3d494a libsystem_kernel.dylib`poll + 10
    frame #0: 0x00007fff8d3d494a libsystem_kernel.dylib`poll + 10
    frame #1: 0x000000010004dca1 libusb-1.0.0.dylib`___lldb_unnamed_function21$$libusb-1.0.0.dylib + 327
    frame #2: 0x000000010004e939 libusb-1.0.0.dylib`libusb_handle_events_timeout_completed + 127
    frame #3: 0x000000010004ea81 libusb-1.0.0.dylib`libusb_handle_events + 34
    frame #4: 0x000000010000b4f8 libuvc.dylib`_uvc_handle_events + 24
    frame #5: 0x00007fff8fca2899 libsystem_pthread.dylib`_pthread_body + 138
    frame #6: 0x00007fff8fca272a libsystem_pthread.dylib`_pthread_start + 137
    frame #7: 0x00007fff8fca6fc9 libsystem_pthread.dylib`thread_start + 13

ktossell, 2015-05-13:
I feel like I've applied lots of fixes for this issue (transfer completion and shutdown processes differ between Linux and Mac, and we are depending on something that only Linux does), and I know the logic is still wrong. I'll see if I can get it fixed soon, now that I have a Mac to try it out with.

saki4510t, 2015-05-13:
Hello,
When I have port libuvc and libusb for Android, libuvc often crash and/or hang-up when stop streaming. I'm not sure it is same issue on OSX because I don't tested on other platform than Android. And my modified version is quite depends on Android specific features and I think pull requests for most of the modifications are not suitable for original libuvc. But I feel my experience may help you.
(Modified version of libusb and uvc are also available on GitHub, `https://github.com/saki4510t/UVCCamera`)

The main cause of crashing when stop streaming on Android is in `uvc_stream_stop` function in `stream.c`.
Original libuvc implementation frees buffers in this function while there are possibilities that other thread (like user callback thread) still use the buffers.
   So I comment-out following lines from `uvc_stream_stop` function.

        // free(strmh->transfers[i]->buffer);
        // libusb_free_transfer(strmh->transfers[i]);
        // strmh->transfers[i] = NULL;

And I also add some code at the end of `_uvc_stream_callback` function;

    if ( strmh->running && resubmit ) {
        libusb_submit_transfer(transfer);
    }  else {
        _uvc_delete_transfer(transfer);
    }

The `_uvc_delete_transfer` function is looks like this;

    static void _uvc_delete_transfer(struct libusb_transfer *transfer) {
        uvc_stream_handle_t *strmh = transfer->user_data;
        if (!strmh) return;
        int i;
        pthread_mutex_lock(&strmh->cb_mutex);
        {
            for (i = 0; i < LIBUVC_NUM_TRANSFER_BUFS; i++) {
                if (strmh->transfers[i] == transfer) {
                    libusb_cancel_transfer(strmh->transfers[i]);
                    UVC_DEBUG("Freeing transfer %d (%p)", i, transfer);
                    free(transfer->buffer);
                    libusb_free_transfer(transfer);
                    strmh->transfers[i] = NULL;
                    break;
                }
            }
            if (i == LIBUVC_NUM_TRANSFER_BUFS) {
                UVC_DEBUG("transfer %p not found; not freeing!", transfer);
            }
            pthread_cond_broadcast(&strmh->cb_cond);
        }
        pthread_mutex_unlock(&strmh->cb_mutex);
    }

I hope this may help you.
regards
saki

mkassner, 2015-05-13:
I m developing on Mac and I experience crashes when using the frame callback. But they never happen when using `uvc_stream_get_frame`.

saki4510t, 2015-05-13:
The behavior is almost same on Android. When using frame callback, many operations execute asynchronously. Transfer buffers are accessed exclusively by `strmh->cb_mutex` on most place, but in `_uvc_user_caller` function, frame callback is called without any exclusive control on line 1020 of stream.c, after unlocking `strmh->cb_mutex` and I assume  sometimes become race condition under some conditions. 

On the other side, `uvc_stream_get_frame` execute synchronously and all functions to use transfer buffers like `uvc_stream_get_frame`, `uvc_stream_stop` etc. exclusivly block by `strmh->cb_mutex` and does not become race condition.

I had tried to lock `strmh->cb_mutex`  when calling frame callback in  `_uvc_user_caller` function, but it did not work well in my case(hang-up/drops lot of frames etc.)

mkassner, 2015-05-18:
working with libuvc I think there are two issues:

1) Occasional crash when using timeouts longer than 2 sec. Not the topic of this issue.

2) Occasional hang on shutdown: As @sadaszewski initially outlined,  uvc_stop_streaming() hangs sometimes. This can be tracked down to `uvc_stream_stop` in stream.c: This loop sometimes never exsits with pthread_cond_wait not returning.

```c
  /* Wait for transfers to complete/cancel */
  do {
    for(i=0; i < LIBUVC_NUM_TRANSFER_BUFS; i++) {
      if(strmh->transfers[i] != NULL)
        break;
    }
    if(i == LIBUVC_NUM_TRANSFER_BUFS )
      break;
    pthread_cond_wait(&strmh->cb_cond, &strmh->cb_mutex);
  } while(1);
```

mkassner, 2015-05-19:
I can confirm the same error on Linux (Ubuntu 14.xx, various kernels).

I have modified `uvc_stream_stop` to use a timeout and return UVC_ERROR_TIMEOUT when appropriate: https://github.com/pupil-labs/libuvc/commit/c534e3d2132351daae4b85e0a6538bc203074dfb

While this is a workaround at best it does do the trick. 

I also found that a failed `uvc_stream_stop` can actually be ignored. I m not sure if this has consequences but the capture can be shut down or restarted after this error event.

rexroni, 2015-08-23:
I have the same problem running Debian Jessie, and the code adjustments offered by saki4510t fix the problem.

rsichel107, 2016-06-20:
hanging error still remains but @saki4510t solutions seemed to fix the problem!

jamesfidell, 2017-07-06:
Since oacapture is mentioned...

I don't see this problem on OSX and haven't done for some time.  In my own fork I've done some clean-up work on a race condition that I think causes trouble when streaming stops.

I'm working on both OSX and Linux with cameras that can generate hundreds of frames per second, so I think I'd still be seeing it somewhere in testing if it were still a problem.