libuvc/libuvc #17: Implement population of capture_time

mkassner, 2015-05-08:
Hi,  looking through the code I noticed that capture_time never get properly populated. IMHO ideally this should be the SOE as in v4l2's uvcvideo module.

Any reason for this? Would this be a lot of work?

I m new to this library and I plan to write a python wrapper similar to this one: https://github.com/pupil-labs/pyv4l2 so that one can get cross platform uvc access from python.

mkassner, 2015-05-11:
I implemented it myself. I ll submit a pull req soon.

mkassner, 2015-05-11:
Actually, this is more work than I expected. Gaining access to `hold_pts` and `dwClockFrequency` was easy and now one has access to the SOE in device time. 

If we want to convert to host time we would have to do what Laurent Pinchart did in the uvc_driver implementation for Linux (http://git.linuxtv.org/cgit.cgi/media_tree.git/tree/drivers/media/usb/uvc/uvc_video.c). This requires knowing the current usb frame number on the host (usb_get_current_frame_number) but this is not part of libusb instead it is platform depended. Alternative implementations for mac/win are available.

Anybody what to join up to implement?

ktossell, 2015-05-13:
That's where I got stuck. I couldn't find any way to get the host-side frame number or other timing information from the host's perspective. I was thinking the timestamp-generation process could fall back to an estimate using a combination of the system timestamp at transfer completion and an estimate of how long it takes for the camera to process and start sending a frame (which it more or less tells you in the device descriptor) and an estimate of how long the transfer would've taken to complete.

mkassner, 2015-05-13:
hi,

I like that. I will submit a pull request for where I m right now. 

BTW.: 
I believe there are functions similar to 'usb_get_current_frame_number' for windows and mac:

https://msdn.microsoft.com/en-us/library/windows/hardware/ff537621(v=vs.85).aspx
https://developer.apple.com/library/mac/documentation/IOKit/Reference/IOUSBDeviceInterface_reference/index.html#//apple_ref/doc/uid/TP40011529-CLSCHIOUSBDeviceInterface-SW8

mkassner, 2015-05-13:
Pull request here: #19

waddlesplash, 2019-07-06:
PR was merged.

closed 2019-07-06 by waddlesplash