libuvc/libuvc #18: Synchronized Capture Across Multiple Devices

lolz0r, 2015-05-11:
Hi,
I would like to utilize many webcams and capture frames from each one in a synchronized fashion. Is this possible to do this with USB wecbams? And if so can you point me in the right direction in the code where actual frame requests / triggers occur?

Thank you for any help you may have!

lolz0r, 2015-05-11:
It looks like the actual trigger to the camera occurs @ libusb_submit_transfer - if I do this in a synchronized way across all camera devices I wonder if this will cause the cameras to take a photo at the same instant?

ktossell, 2015-05-13:
I don't know of a way to do that. I didn't think that the cameras actually synchronized themselves with the USB transfers. I thought they maintained their own schedule and simply report the result if/when it's ready. Maybe that's device-dependent. There's currently no allowance for multi-camera interaction in libuvc; I don't have a good place to point to where you could insert synchronization code. Though if you want, you could try using two separate libuvc contexts and hacking in some mutex-lock-wrapped code immediately before the `libusb_submit_transfer` calls in the stream callback. (Make a global pthread_mutex_t and pthread_cond_t. Initialize in your code. If handling camera 1 and you get the end-of-frame transfer result, update a counter with the lock held and signal a condition variable. If handling camera 2 and you get and end-of-frame, check the counter with the lock held, and if it hasn't been updated, wait on the condition variable before resubmitting your transfer.) You'd need separate UVC contexts because the worker thread is shared among all cameras in a context, and if you just had one thread, you'd get stuck forever waiting for the condition variable to be signaled.

closed 2017-02-23 by lolz0r