libuvc/libuvc #21: Shouldn't struct uvc_format_desc have a member enum uvc_frame_format?

olafkaehler, 2015-06-24:
Maybe I'm being silly here, but shouldn't the struct uvc_format_desc have a member enum uvc_frame_format? I'm trying to implement a function like "find me a suitable format for this and that resolution". The return values of this function should then be the input to uvc_get_stream_ctrl_format_size().

I've found the function uvc_get_format_descs() (which is not in the doxygen documentation, btw.) to get me a list of formats and resolutions available for a camera, but I can't map the GUID in its return values to the enum uvc_frame_format required by uvc_get_stream_ctrl_format_size.

What am I missing?

ktossell, 2015-06-30:
But there isn't always exactly one uvc_frame_format for each format GUID. I agree that, with the uvc_format_desc and uvc_frame_desc, you should have everything you need to open a stream, and you shouldn't need to map the guid back to a uvc_frame_format. I'd rather resolve the lack of that support by adding a function along the lines of `uvc_get_stream_ctrl_for_frame_desc(uvc_device_handle_t *devh, uvc_stream_ctrl_t *ctrl, const uvc_frame_desc *desc)`

olafkaehler, 2015-06-30:
Sounds sensible. Maybe also add a function for mapping the "interval" to the fps, as required by uvc_get_stream_ctrl_format_size().

olafkaehler, 2015-10-27:
I've added the discussed function in my local fork. Feel free to merge it back to the master if you find it useful.