libuvc/libuvc #23: Question: Force bulk transfer

mkassner, 2015-06-30:
Hi,

I m not sure if this is possible. Some cameras have alt settings with tremendously oversized packet sizes.

Is is possible to force bulk transfer?
(I tried forcing bulk transfers in stream.c but  libusb_submit_transfer fails with 'Entity not found.' .)

Any pointers would be very appreciated.

ktossell, 2015-07-10:
I believe that UVC camera streaming interfaces can do isochronous transfers or bulk transfers, but not both. The camera would have to provide more than one streaming interface for that to be possible.

I think the approach you're taking in https://github.com/pupil-labs/libuvc/commits/bandwidth_fix is the best way to go.

mkassner, 2015-07-20:
Alright that makes sense. Closed

closed 2015-07-20 by mkassner