libuvc/libuvc #42: UVC 1.5

Pazu, 2016-03-03:
Is version 1.5 of the class specification already supported or, if not, is it planned? At present it appears only version 1.1. is supported; is that right?

Pazu, 2017-06-06:
Could someone please comment on this? Is UVC 1.5 alive, with even the slightest prospect for support on Linux?

jamesfidell, 2017-07-06:
Currently I think 1.0 and 1.1 are "mostly" supported.  I don't think anything new in 1.5 has been added at all.  You may need to find someone who wants to scratch that particular itch if you need it doing :)

sabetAI, 2018-08-12:
How about now?

seker, 2019-06-25:
What a pity! 
it still don't support UVC 1.5, I am hopeless~

waddlesplash, 2019-07-06:
Is there some specific camera you can point to that uses UVC 1.5 which UVC 1.1 does not work on? Or is there some UVC 1.5 feature you really need to access?

Croydon, 2019-07-06:
The Brio Ultra HD Pro Webcam doesn't seem to work with UVC 1.1

hr0109, 2020-06-23:
> Is there some specific camera you can point to that uses UVC 1.5 which UVC 1.1 does not work on? Or is there some UVC 1.5 feature you really need to access?

ricoh theta v
Logitech 930e

hr0109, 2020-06-23:
> Is there some specific camera you can point to that uses UVC 1.5 which UVC 1.1 does not work on? Or is there some UVC 1.5 feature you really need to access?

More and more webcam are using uvc 1.5 with payload h.264

closed 2020-07-19 by waddlesplash

traud, 2021-02-17:
@hr0109 and @Croydon out of curiosity, does the attached change help with your mentioned Logitech cameras? I am using the Logitech C922 which is known to [hide its H.264 stream …](https://raspberrypi.stackexchange.com/a/91885) After building and installing the latest trunk, my camera still does not show any H.264 stream. Forcing it to `UVC_FRAME_FORMAT_H264` fails as well. I guess, this is just for those camera listing their H.264 stream like the original Logitech C920. Right?

Croydon, 2021-02-18:
I can't test this currently

Pazu, 2021-02-20:
I could try to test it with my c930e. Is trunk necessary or is the patch in 0.0.6?

traud, 2021-02-20:
You need trunk.

dalbani, 2021-12-17:
> I could try to test it with my c930e. Is trunk necessary or is the patch in 0.0.6?

@traud: did you try by any chance with a C930e?
Well, actually, I have and I didn't see any difference in the reported list of video formats.
I am wondering if this particular camera requires some particular trickery.
As shown in a Windows / DirectShow environment by having to use an alternative "capture pin":
https://stackoverflow.com/questions/24675538/how-to-support-hardware-encoded-h264-though-uvc
https://stackoverflow.com/questions/25086843/how-to-access-the-h264-stream-from-webcams-that-feature-h264-hardware-encoders/33258812

The C930e is also supposed to support SVC.
That reminds me that while browsing the UVC 1.5 standard and its H.264 payload documentation, I've also come across this bit:
> When supported, the use of SVC and simulcast of multiple streams in the context of this
specification shall additionally conform to the following specification (hereafter referred
to as UCConfig specification):
a. Unified Communication Specification for H.264/MPEG-4 AVC and SVC Encoder
Implementation. This specification is available at
http://technet.microsoft.com/en-us/lync/gg278176.aspx. Unless otherwise
specified, this document refers to the edition of version 1.1 published in April
2011 (posted at the Microsoft web site link above).

Which actually points to this document: http://download.microsoft.com/download/A/F/B/AFBF8CBE-3A45-472A-93F3-AD8521FBD502/UC_Specification_for_H264_AVC_and_SVC_encoder.pdf

So is the puzzle even more complicated with particular webcam?

dalbani, 2021-12-17:
By the way, I just mentioned SVC *1.5* but the capabilities exposed by the device seem to only mention UVC 1.0 strangely enough?!? 😕 
On Linux but also on Windows as seen in the output of USBTree: https://gist.github.com/dalbani/3fa6621c09baff3d94cc.
But I'm pretty sure it supports UVC 1.5, as mentioned in the specs, and also as seen as in a firmware dump: https://gist.github.com/dalbani/12e017fdec447796d798.

traud, 2021-12-21:
I do not own a C930e myself. I think you were about mentioning @Pazu.

dalbani, 2021-12-21:
Thanks @traud, I pinged the wrong person indeed.
Should someone (ever) be interested, I've posted the output of an (H264 enhanced) `lsusb` command for my Logitech C930e on Linux: <https://gist.github.com/dalbani/08f295fe436b76fa70bd29e1f51321cc>.
I actually had to use a hack to make the second configuration descriptor visible to the Linux USB stack: <https://sourceforge.net/p/linux-uvc/mailman/message/37404699/>.

Next step is to try to extract H264 data from those endpoints.
On the condition that I first manage to understand all the technical documentation for UVC / H264 / SVC... 😜

jujurou642, 2022-07-13:
@dalbani : Would you please publish the (H264 extended) lsusb command for Logitech C930e?

jujurou642, 2022-07-14:
@dalbani : I have changed the kernel and got the second Configurations with the lsusb command. However, since the lsusb command does not support UVC1.5, I would like to see the source code of the lsusb command you modified.

jujurou642, 2022-07-27:
I created a patch that just looks at the descriptor information with the lsusb command.

kernel_usb_c930e.patch
https://1drv.ms/u/s!AncVN32qssoflXNZDqsSSOwJrMZE?e=PaQJaY

usbutils_20220719_uvc1.5_h264.patch
https://1drv.ms/u/s!AncVN32qssoflXRjwsJVnr_Mscud?e=a9ADtU

dalbani, 2022-07-28:
Hi @jujurou642, sorry for the delay getting back to you. I had to dig into my files to find this diff for `lsusb`:

```
diff --git a/lsusb.c b/lsusb.c
index 46593a0..0eda0ef 100644
--- a/lsusb.c
+++ b/lsusb.c
@@ -2006,6 +2006,200 @@ static void dump_videostreaming_interface(const unsigned char *buf)
 		dump_junk(buf, "        ", 24);
 		break;
 
+	case 0x13: /* FORMAT_H264 */
+		printf("(FORMAT_H264)\n");
+		if (buf[0] != 52)
+			printf("      Warning: Descriptor too short\n");
+
+		printf("        bFormatIndex                    %5u\n"
+		       "        bNumFrameDescriptors            %5u\n"
+		       "        bDefaultFrameIndex              %5u\n"
+		       "        bMaxCodecConfigDelay            %5u\n",
+		       buf[3], buf[4], buf[5], buf[6]);
+		printf("        bmSupportedSliceModes            0x%02x\n",
+		       buf[7]);
+		// TODO explain bitmap
+		printf("        bmSupportedSyncFrameTypes        0x%02x\n",
+		       buf[8]);
+		// TODO explain bitmap
+		printf("        bResolutionScaling              %5u ",
+		       buf[9]);
+		switch (buf[9]) {
+		case 0:
+			printf("(Not supported)\n");
+			break;
+		case 1:
+			printf("(Limited to 1.5 or 2.0 scaling in both directions)\n");
+			break;
+		case 2:
+			printf("(Limited to 1.0, 1.5 or 2.0 scaling in either direction)\n");
+			break;
+		case 3:
+			printf("(Limited to resolutions reported by associated Frame Descriptors)\n");
+			break;
+		case 4:
+			printf("(Arbitrary scaling)\n");
+			break;
+		default:
+			printf("(Reserved)\n");
+		}
+
+		printf("        bmSupportedRateControlModes      0x%02x\n" // TODO explain bitmap
+		       "        wMaxMBperSecOneResolutionNoScalability                  %5u\n"
+		       "        wMaxMBperSecTwoResolutionNoScalability                  %5u\n"
+		       "        wMaxMBperSecThreeResolutionNoScalability                %5u\n"
+		       "        wMaxMBperSecFourResolutionNoScalability                 %5u\n"
+		       "        wMaxMBperSecOneResolutionTemporalScalability            %5u\n"
+		       "        wMaxMBperSecTwoResolutionTemporalScalability            %5u\n"
+		       "        wMaxMBperSecThreeResolutionTemporalScalability          %5u\n"
+		       "        wMaxMBperSecFourResolutionTemporalScalability           %5u\n"
+		       "        wMaxMBperSecOneResolutionTemporalQualityScalability     %5u\n"
+		       "        wMaxMBperSecTwoResolutionTemporalQualityScalability     %5u\n"
+		       "        wMaxMBperSecThreeResolutionTemporalQualityScalability   %5u\n"
+		       "        wMaxMBperSecFourResolutionTemporalQualityScalability    %5u\n"
+		       "        wMaxMBperSecOneResolutionTemporalSpatialScalability     %5u\n"
+		       "        wMaxMBperSecTwoResolutionTemporalSpatialScalability     %5u\n"
+		       "        wMaxMBperSecThreeResolutionTemporalSpatialScalability   %5u\n"
+		       "        wMaxMBperSecFourResolutionTemporalSpatialScalability    %5u\n"
+		       "        wMaxMBperSecOneResolutionFullScalability                %5u\n"
+		       "        wMaxMBperSecTwoResolutionFullScalability                %5u\n"
+		       "        wMaxMBperSecThreeResolutionFullScalability              %5u\n"
+		       "        wMaxMBperSecFourResolutionFullScalability               %5u\n",
[... 114 lines cut ...]
+		       mvc_capabilities);
+		// TODO explain bitmap
+
+		printf("        dwMinBitRate                 %8u\n"
+		       "        dwMaxBitRate                 %8u\n"
+		       "        dwDefaultFrameInterval         %5u\n",
+		       buf[31] | buf[32] << 8 | buf[33] << 16 | buf[34] << 24,
+		       buf[35] | buf[36] << 8 | buf[37] << 16 | buf[38] << 24,
+		       buf[39] | buf[40] << 8 | buf[41] << 16 | buf[42] << 24);
+
+		uint num_frame_intervals = buf[43];
+		printf("        bNumFrameIntervals              %5u\n",
+		       num_frame_intervals);
+		for (i = 0; i < num_frame_intervals; i++)
+			printf("        dwFrameInterval(%2u)         %9u\n",
+			       i, buf[44+4*i] | (buf[44+4*i+1] << 8) |
+			       (buf[44+4*i+2] << 16) | (buf[44+4*i+3] << 24));
+		// ...
+		// dump_junk(buf, "        ", 11); ???
+		break;
+
 	default:
 		printf("        Invalid desc subtype:");
 		dump_bytes(buf+3, buf[0]-3);
```

skmagiik, 2022-09-15:
Obviously, we still don't have a UVC 1.5 solution. What is the best way to get someone to work on this?

PR #143 (2020-07-19): Feature/h264
uvc 1.5 with payload h.264 close #42 #66