libuvc/libuvc #50: example.c ends in a segfault

rsichel107, 2016-06-10:
I am trying to use a logitech camera for an application I am working on. I am curious if anyone else has successfully run example.c. 

I downloaded the code as instructed on my linux machine. I moved the example.c code into the build directory and compiled it with this:
`gcc  example.c -o exampleprog -luvc`
(Originally there was an access denied problem so I am running it as a `root` user, but that does not seem be the issue. )

When running the code ti ends in,` Segmentation fault (core dumped)`. I have isolated the problem to be in the callback function- specifically in the while loop of the `uvc_yuyv2bgr()`` function. 

Any advice or solutions for this?

Thank you-
RLS

coreylowman, 2016-09-28:
I am currently running it on a beaglebone black, and sometimes it runs fine, but sometimes it does segfault. I ran gdb on it and it seems like its crashing in a call libusb. It crashes in uvc_stop_streaming call.

ovchinnikov-v-a, 2016-10-19:
It seems I've had the same trouble - my app using libuvc crashes when trying to close device with uvc_close() after starting a stream with uvc_stream_start().

The actual problem was in function called uvc_stream_stop - there is erroneous usage of libusb_cancel_transfer return value:

`int res = libusb_cancel_transfer(strmh->transfers[i]);
if(res < 0) {
 free(strmh->transfers[i]->buffer);
 libusb_free_transfer(strmh->transfers[i]);
 strmh->transfers[i] = NULL;
}`

It is not valid to free transfer immediately after libusb_cancel_transfer returned negative result cause this transfer may still be used in uvc_iso_callback function.

So I commented out transfer freeing code in uvc_stream_stop and add transfer deletion to uvc_iso_callback for LIBUSB_TRANSFER_COMPLETED, LIBUSB_TRANSFER_TIMED_OUT, LIBUSB_TRANSFER_STALL and LIBUSB_TRANSFER_OVERFLOW cases.
Of course transfer deletion in uvc_iso_callback for these cases should happen only when strmh->running == 0.

I'm not sure if this quick and dirty solution is the best one, but it works for me - no more crashes when calling uvc_close(). My current version of stream.c is attached.
[stream.zip](https://github.com/ktossell/libuvc/files/539594/stream.zip)