libuvc/libuvc #53: sometimes program hangs when uvc_stop_streaming(devh) is called (even with test.c)

amitv87, 2016-08-24:
Tried with different cams and it happens with all even on different operating systems (os x, ubuntu and raspbian)

ovchinnikov-v-a, 2016-10-19:
Maybe solution from #16 or #50 will help

hipersayanX, 2016-12-31:
I can confirm that this issue can be solved applying the changes from [#16](https://github.com/ktossell/libuvc/issues/16#issuecomment-101653441).

stevenlovegrove, 2017-03-10:
Just a +1 from me on this issue on Ubuntu 16.04.

Similar it seems to other users, I traced the problem down to the cancel transfer logic, where each LIBUSB_ERROR_NOT_FOUND results in strmh->transfers[i] not being set to null
https://github.com/ktossell/libuvc/blob/master/src/stream.c#L1196

and hence an infinite loop here:
https://github.com/ktossell/libuvc/blob/master/src/stream.c#L1207

My workaround is to accept a memory leak and set strmh->transfers[i] to null for every error case, and only deallocate when not LIBUSB_ERROR_NOT_FOUND.

littlewater, 2019-08-19:
the same error on my linux pi system, this error never occured on my windows 10.
After try with DEBUG flag on, it tells me (on linux sys), when USB line error (you could plug out for easy reproduce the problem ), some transfer object won't be freed due to the driver impl, 
if as the original code  res = libusb_cancel_transfer with return value res = LIBUSB_ERROR_NOT_FOUND, the code just SKIP the free, but as the doc of libusb, it says, NOT FOUND means, already completed, already canceled or not submit state.

thus I think the code should NOT handle if (res < 0 && res != LIBUSB_ERROR_NOT_FOUND),but should be if (res < 0 && res == LIBUSB_ERROR_NOT_FOUND).

And I retry with my update code, this hangs won't be reproduced, I think it should be a code bug.

netman69, 2022-03-06:
I suspect this is the same issue as #152, I am working on a PR

mcuee, 2022-05-08:
https://github.com/libuvc/libuvc/pull/210 has been merged. It may have aready fixed this issue.

pandeypradyumn, 2022-06-30:
any possible fixes for windows ?