libuvc/libuvc #64: Bulk Transfer results in LIBUSB_TRANSFER_TIMED_OUT

ojotoxy, 2017-02-02:
Occasionally upon connecting to the device and opening a stream, I only get LIBUSB_TRANSFER_TIMED_OUT inside _uvc_stream_callback.

I can only make this go away by reconnecting the device.

My device has a blinking LED when the transfer is working, and it's not blinking in this state.

Any ideas why this would happen?

This might not really be a libuvc issue, it might be bugginess in the hardware or Android OS.
I am acutally using https://github.com/saki4510t/UVCCamera which has modified this library.

waddlesplash, 2019-07-06:
Is it still reproducible, even on Linux with an unmodified libuvc?

Eddielee30, 2021-04-14:
Hi, would you try to change the LIBUVC_NUM_TRANSFER_BUFS to smaller to check will it still happen

jiaozebo, 2022-02-23:
> 

to change this value did work. could you please explain what this value means, why smaller is good?  what's the proper default value