libuvc/libuvc #70: Trouble installing libuvc

AnthonyTheKoala, 2017-05-22:
I have tried to install libuvc. I downloaded libuvc-master. 
Using the GUI version of CMake, I am still getting the following errors.
<pre>
The C compiler identification is MSVC 19.0.24215.1
The CXX compiler identification is MSVC 19.0.24215.1
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
No build type selected, default to Release
No target type selected, default to shared library
Building libuvc with JPEG support.
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
JPEG_LIBRARY_NAMES
    linked by target "uvc" in directory C:/Python34/libuvc-master
LIBUSB_INCLUDE_DIR
   used as include directory in directory C:/Python34/libuvc-master
   used as include directory in directory C:/Python34/libuvc-master
LIBUSB_LIBRARY_NAMES
    linked by target "uvc" in directory C:/Python34/libuvc-master
PTHREAD_INCLUDE_DIR
   used as include directory in directory C:/Python34/libuvc-master
   used as include directory in directory C:/Python34/libuvc-master
PTHREAD_LIBRARY_NAMES
    linked by target "uvc" in directory C:/Python34/libuvc-master

Configuring incomplete, errors occurred!
See also "C:/Python34/libusb-master/CMakeFiles/CMakeOutput.log".
</pre>
What are these values? Are they environment values? Where are they supposed to be set?
<pre>
JPEG_LIBRARY_NAMES, LIBUSB_INCLUDE_DIR, PTHREAD_INCLUDE_DIR, PTHREAD_LIBRARY_NAMES
</pre>

The C:/Python34/libusb-master/CMakeFiles/CMakeOutput.log is attached. But there is nothing extraordinary as there is no evidence of errors.
[cmakeoutput.log.txt](https://github.com/ktossell/libuvc/files/1020170/cmakeoutput.log.txt)

My environment:
<pre>
OS: Windows 7, 64-bit
CMake: 3.8.0rc2
Compiler: MSVC 2015 version 19.0.24215.1 64-bit
Python: 3.4, 64-bit
</pre>

I am just one step closer to getting a successful compile. Any assistance would be grateful.

Thank you,
Anthony, Sydney NSW

stevecotton, 2017-05-31:
Hi Anthony,

The user who wrote the Win 8 + Python 3.5 HOWTO that you linked to in issue #69 also has the large pull request #68, where he/she says adding Windows support will require work to apply. As you're trying to get it working with Win 7 and Python 3.4, I'd assume the task is development and porting, not just installing; it's going to be harder.

The bit about `JPEG_LIBRARY_NAMES, LIBUSB_INCLUDE_DIR, PTHREAD_INCLUDE_DIR, PTHREAD_LIBRARY_NAMES` means that CMake's find_package function can't find libraries and headers for libjpeg and pthreads. But before you start learning to debug CMake makefiles on a complex project which will need some porting and will probably need debugging of threaded I/O before it works, I think you should consider whether switching to a different platform for this project would be simpler.

Steve

AnthonyTheKoala, 2017-05-31:
Dear Steve,

Thank you for your reply it is appreciated.

Is there anything of a ‘lower abstraction’, that is below libusb’s abstraction which will allow me to control a UVC camera, that is a USB camera that in addition to setting and getting the resolution and setting and getting the audio will also allow me to control the pan/tilt/zoom functions.

 

I know for wireless webcams that “http://”  type commands are used for setting/getting , but “http://” is not used.  There is ‘plenty’ of materials on wireless webcams controlled by python, but there is very little on USB/UVC.  

 

NOTE: I know that OpenCV allows wireless or USB/UVC access to cameras, but that is only for acquiring images to be used for image processing. As a result there is nothing to control the audio and the pan/tilt/zoom of USB/UVC cameras.

 

I have successfully installed both libusb and pyaudio. But don’t know how to get audio from a USB/UVC camera and have no idea how to control pan/tilt/zoom.

 

Therefore if libuvc cannot be used are there any packages you can suggest any other access to a USB/UVC camera.

 

Thanks, merci, danke,

Anthony, Sydney Australia

s-eswar, 2018-02-19:
Hi People, 
I'm using this libuvc for reading input from my usb3.0 stereo camera(from leopard imaging). I installed the libuvc. However, when I run the ./uvc_test, I'm unable to see any display(any capture) from the camera. A blank white screen appears...
what to do?

waddlesplash, 2019-07-06:
That is a different issue; please open a separate ticket.

Windows support is also tracked elsewhere, so closing this.

closed 2019-07-06 by waddlesplash