libuvc/libuvc #72: Multi camera support

caoyangjiang, 2017-05-31:
I got single camera to work. I am trying to get two cameras to stream at the same time. I tried creating one context two devices, two context one for each device, and moving two devices onto two different threads. It seems none of these works. The second device will never pass uvc_open() although the first one works just fine. Does this library support multiple devices simultaneous streaming? If so, what is the proper way to do it? Thank~!

stevecotton, 2017-05-31:
What errors are you getting from uvc_open()?

Are both cameras on the same USB root hub, and does it have enough bandwidth for both cameras?

caoyangjiang, 2017-05-31:
I did not get any error from uvc_open(). It can't proceed to the next line of code, which prints the error. Both cameras are directly plugged into the USB ports on the backside of the computer. I think there should not be any competition on the bandwidth. What is the correct procedure to handle multiple cameras? Should I create one context or two context ?

caoyangjiang, 2017-06-01:
I experimented again with single context and two uvc_open(). Both camera device open normally. However, if I call uvc_start_streaming() for one device, it will block uvc_start_streaming() for the second device. The program stuck on the second uvc_start_streaming(). No callback is called for the second uvc_start_streaming().

matejjeglic, 2019-05-16:
Has any progress been made regarding this issue. I am experiencing the same problem.

ArtlyStyles, 2019-06-21:
Anyone fixed this problem?

jwlodek, 2019-06-22:
I have written a libuvc based driver for scientific image processing (https://github.com/jwlodek/ADUVC), and I have experimented quite a bit with running multiple cameras off of one server. I found that if the cameras had an identical product ID, they would block each other from streaming simultaneously. In addition, in certain cases, depending on the USB interface they are connected to, and the way it is structured on your motherboard, the startstreaming calls may block. Unfortunately I have not yet been able to find a solution to this other than ensuring the cameras are on different USB interfaces on the host server

nwohaibi, 2019-12-02:
for what it's worth, see [this](https://raspberrypi.stackexchange.com/questions/105358/raspberry-pi4-error-while-using-2-usb-cameras-vidioc-qbuf-invalid-argument) and [this](https://superuser.com/questions/431759/using-multiple-usb-webcams-in-linux).

Parrotlife, 2020-04-23:
Hello everyone, I am facing the same issue for running 2 cameras at the same time as they have the same product ID. If someone found a turnaround for this problem that would be great :) 
Also @jwlodek how can you setup the 2 cameras on different USB interfaces? 
I am running this with ubuntu 18.04 on an nvidia jetson tx2.
Thank you all for your answers ;)

jwlodek, 2020-04-23:
@Parrotlife For the product ID issue the different interfaces won't fix the problem, you can try using serial number to connect instead, that would help. The interface solution fixes an issue where sometimes cameras even with different product IDs would block themselves when running off the same USB bus controller on the motherboard, though I can't reliably reproduce this behavior

Parrotlife, 2020-04-23:
@jwlodek Thank you for your answer. My cameras don't have any serial number. can I somehow force one on them? Thanks again for your help.

jwlodek, 2020-04-23:
@Parrotlife Not that I know of unfortunately

Parrotlife, 2020-04-24:
I managed to find my serial numbers using "sudo lsusb -v" (doesn't work without sudo) and then I pass this serial number to find_uvc_device. This allowed me to open both cameras. Thanks @jwlodek for your help.

bookshiyi, 2020-06-19:
one "common hub" : one camera work same time.
one "root hub " : multi camera work same time.

this is my test result

suasllc, 2020-08-06:
> I managed to find my serial numbers using "sudo lsusb -v" (doesn't work without sudo) and then I pass this serial number to find_uvc_device. This allowed me to open both cameras. Thanks @jwlodek for your help.

This works for me. Thanks a lot for your help!
I think the reason is that when you have 2 or more cameras of the same brand, their vid and pid are the same. So you will need the third parameter being the serial #, which differ from camera to camera, to specify a specific one. Without this, find_uvc_device will only (probably) stop at the first found camera, which in the case of multi-cam situation, will result in a busy (-6) error for subsequent opening, because it tries to open the same first-found camera, but not the next found ones.

rjgebis, 2021-02-18:
@suasllc I have same issue with two MS cameras
```
Bus 003 Device 007: ID 045e:075d Microsoft Corp. LifeCam Cinema
Bus 003 Device 007: ID 045e:075d Microsoft Corp. LifeCam Cinema
```
But when I do `lsusb -v ` I see both cameras have iSerial 0
```
 $ lsusb -v -d 045e:075d | grep iSerial
Couldn't open device, some information will be missing
  iSerial                 0 
```

@waddlesplash 
I am not familiar with libusb. Is there any way to make this work? I could try to implement and do PR

bookshiyi, 2021-02-18:
After our test, we found that the reason why multiple cameras can't be opened at the same time is caused by insufficient usb bandwidth. My conclusion is that you can try to choose two usb camera that supports lower bitrate and try after configuring it to the lowest bitrate.

rjgebis, 2021-02-24:
@bookshiyi Do you have a working sample of how to select two identical cameras with no iSerial so I can try it?

ccpearson, 2021-07-29:
> @bookshiyi Do you have a working sample of how to select two identical cameras with no iSerial so I can try it?

I don't have cameras with identical serial numbers to test with right now, but it seems like you could do it with uvc_get_device_list(), which doesn't use serial number to find the camera. (Your code would then have to use some other way to identify the cameras, like array index or something.)

```
#define MAX_CAMERAS 5 // or whatever

uvc_device_handle_t *handles[MAX_CAMERAS] = { 0 };
int num_cameras = 0;

void open_all_uvc_cameras() {

    uvc_context_t *ctx;
    uvc_error_t err = uvc_init(&ctx, NULL);

    if (err == UVC_SUCCESS) {
        uvc_device_t **list;
        err = uvc_get_device_list(ctx, &list);

        if (err == UVC_SUCCESS) {
            for (num_cameras = 0; num_cameras < MAX_CAMERAS && list[num_cameras]; ++num_cameras) {

                uvc_device_descriptor_t *desc;
                err = uvc_get_device_descriptor(list[num_cameras], &desc);

                if (err == UVC_SUCCESS) {
                    err = uvc_open(list[num_cameras], &handles[num_cameras]);
                    uvc_free_device_descriptor(desc);
                }
            }
            uvc_free_device_list(list, true);
        }
    }
}
```
Great thread BTW, this will save us a lot of headaches in the field.

tonyngophd, 2021-07-30:
How about this: 

https://github.com/tonyngophd/vuir-zoom/blob/master/src/uvcacquisition.cpp

We solved the issue with multiple FLIR boson camera:

https://youtu.be/d1FKRNGz8qg

Thanks

ccpearson, 2021-07-30:
> We solved the issue with multiple FLIR boson camera:

Impressive! But I was referring to the issue of the libuvc hang that occurs in uvc_find_device() when there are cameras having duplicate serial numbers. Your code uses uvc_find_device() rather than uvc_get_device_list(). So I don't think we're talking about the same issue.

bookshiyi, 2021-08-01:
> @bookshiyi Do you have a working sample of how to select two identical cameras with no iSerial so I can try it?

> I don't have cameras with identical serial numbers to test with right now, but it seems like you could do it with uvc_get_device_list(), which doesn't use serial number to find the camera. (Your [...]

@ccpearson @rjgebis 

As far as I know, most usb cameras are implemented based on the CSI protocol, and the CSI protocol also includes IIC. If the manufacturer does not deliberately distinguish the IIC address, then cameras of the same brand and model may not be so easy to distinguish. 

BTW, you can fix the relationship between the USB interface and the camera to achieve your goal.

Breezewrf, 2024-08-29:
> I managed to find my serial numbers using "sudo lsusb -v" (doesn't work without sudo) and then I pass this serial number to find_uvc_device. This allowed me to open both cameras. Thanks @jwlodek for your help.

Additional: need to wrap the serial number with 'b' like: `b'5e4315465ef1wf455'`