libuvc/libuvc #73: Need help with call back function

asifbashir, 2017-06-01:
Hi, 
  I am new to c and libuvc. I successfully run example.c and confused about call back function. 
Actually i need to maintain a buffer , lets say of size 1000 frames, and whenever a user press a key I have to create a video of this buffer . Can someone please guide how can i achieve this ?

caoyangjiang, 2017-06-02:
You can declare a member variable to hold 1000 frames. In that case, you don't have to "create" buffer every time. You can simply overwrite this buffer. The callback function is always running once you call uvc_start_streaming(). So I suggest you create an atomic variable of boolean type and a counter. Once the user press a key, you set counter to 0 and then boolean variable to true (order matters!). In the callback function, once this boolean variable becomes true you start to fill in the buffer and increment the count. Once the count becomes 1000, inside callback function, set the atomic variable to false. On the client side, you can either check the count or the boolean variable to decide whether 1000 frames has been captured. If you decide to check count, you must also declare it as atomic variable. Hope this helps.

asifbashir, 2017-06-05:
Hi, Thanks for your help. I have a buffer containing frames, Now I want to write a video (mp4 or avi) file. How can I do this ? Need some command or code .

caoyangjiang, 2017-06-13:
That is a whole another story. I'd suggest doing it with command. Before you turn frames into mp4, you want to compress it with H.264 or H.265. You can save all the frames in RGB or YUV format and then use ffmpeg to concatenate, compress and encapsulated inside a mp4 or avi file.

closed 2019-07-06 by waddlesplash