libuvc/libuvc #74: asynchronous/callback vs synchronous/polling

lobsteroh, 2017-06-28:
i am getting great performance out of the library for streaming (i think i am using synchronous) and i am now trying to push it closer to the framerates my high-speed camera is capable of (according to specs). i would love to get frames more often (even if they are duplicates of previous ones). anyone have experience with grabbing frames in asynchronous mode? any examples much obliged.

jamesfidell, 2017-07-06:
I don't know what sort of frame rate you're talking about, but I have a fork of libuvc that I've used for frame rates of hundreds per second using asynchronous transfers.  I'm hoping Ken will pull the changes I've made back into the main repo.

waddlesplash, 2019-07-06:
Appears @jamesfidell's changes have been merged :)

closed 2019-07-06 by waddlesplash