libuvc/libuvc #83: CMake Error: install TARGETS given target "uvc_static" which does not exist

reagentoo, 2017-08-03:
```
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: x86_64-pc-linux-gnu-pkg-config (found version "0.29.2") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.21
-- Checking for module 'libjpeg'
--   Found libjpeg, version 1.5.2
-- Building libuvc with JPEG support.
CMake Error at CMakeLists.txt:95 (install):
  install TARGETS given target "uvc_static" which does not exist in this
  directory.

-- <<< Gentoo configuration >>>
Build type      Gentoo
Install path    /usr
Compiler flags:
C               -march=core-avx2 -O2 -pipe
C++             -march=core-avx2 -O2 -pipe
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed
Module          -Wl,-O1 -Wl,--as-needed
Shared          -Wl,-O1 -Wl,--as-needed

-- Configuring incomplete, errors occurred!
```

ktossell, 2017-08-03:
Could you please try again with the changes I just pushed? Shared-only and static-only builds were broken.

reagentoo, 2017-08-03:
It seems everything ok. Thanks.

closed 2019-07-06 by waddlesplash