libuvc/libuvc #85: uvc_get_input_select crashes

avilleret, 2017-08-20:
both on Mac OS X and Ubuntu, I got a crash when calling `uvc_get_input_select` with different cameras.

here are the crashlog and the `example.c` to reproduce the crash : https://gist.github.com/avilleret/4feffddc9c95bcf075b1dfd909b12ce6

moreover I tried with `UVC_GET_CUR` and `UVC_GET_INFO` request

any idea on what is wrong here ?

openastroproject, 2017-08-20:
A quick scan suggests that no selector units have been parsed so presumably the camera doesn't have any and a call to "uvc_get_input_select()" therefore makes no sense.

Have you tried calling "uvc_get_selector_units()" first to see if it returns a non-null value?

avilleret, 2017-08-20:
ha OK that makes sense, I was thinking that `uvc_get_input_select(m_devh, &input_select, UVC_GET_INFO)` will return a non capable flag and `uvc_get_input_select(m_devh, &input_select, UVC_GET_CUR)` and error if the device doesn't have this capability

anyway, it seems to be fixed now
thanks for the quick reply and congrats for all the work

closed 2017-08-20 by avilleret