libuvc/libuvc #86: request to add new tag v0.0.6 ( or v0.1.0 ?)

k-okada, 2017-09-21:
hi @ktossell 
can you add new tag to the source tree? since latest version was v0.0.5 and that was 3 years ago.
we have requests from  ROS users who want to use latest features of libuvc.
Of course, if you add tag information, I can do all jobs related to releasing into ROS ecosystem 

Cc: @7675t

ktossell, 2017-09-21:
just pushed https://github.com/ktossell/libuvc/tree/v0.0.6 :smile:

k-okada, 2017-09-21:
Thank you for quick response!!!
I have updated the changelog, without understanding what each commit really means, hope this could be good starting point for your edit....
https://github.com/ktossell/libuvc/pull/87

k-okada, 2017-09-21:
@ktossell could you add @7575t and @gbiggs to these release packages? They'll also help us to release into ros build firm...

https://github.com/ktossell/libuvc-release
https://github.com/ktossell/libuvc_ros-release
https://github.com/ktossell/camera_umd-release

ktossell, 2017-09-21:
I think I lost the ability to change repo permissions when I transferred those repos to ros-drivers-gbp. @jack-oquin or a ros-drivers-gbp maintainer may be able to add @7675t to the camera_drivers team, which has access to those repos.

7675t, 2017-09-22:
@ktossell hi, can I say hello to you or you are not here ROSCon?

closed 2019-07-06 by waddlesplash