libuvc/libuvc #91: Capture video and data from device

nussetorten, 2017-11-16:
I have an interest in streaming video data and bulk data from the same USB3.0 device.  My current implementation features a libuvc callback to retrieve video frames and a libusb_bulk_transfer to retrieve data.  The bulk transfer happens in a separate thread from the video transfer.

My initial implementation succeeds in retrieving data and video but demonstrates some instabilities, especially around destruction time.  libuvc hangs while cancelling some of the 100 transfers.  I suspect that libusb-side thread safety is to blame.

**The big ask**: do you have any insight for stable simultaneous video and bulk transfer?  Would love to see a wiki page or something result from this.

_Consulted libuvc documentation, and this page on multithreading & libusb: http://libusb.sourceforge.net/api-1.0/mtasync.html_

nussetorten, 2017-11-16:
The hang appears to occur when libusb_cancel_transfer gives a return code of LIBUSB_ERROR_NOT_FOUND (L1231):

https://github.com/ktossell/libuvc/blob/77b43d618e71698f52f7beb88c4b99b5f018259b/src/stream.c#L1228-L1237

The following cond wait never resolves:

https://github.com/ktossell/libuvc/blob/77b43d618e71698f52f7beb88c4b99b5f018259b/src/stream.c#L1247

nussetorten, 2017-11-17:
I've authored a short application which triggers the error.  Funny enough, I don't event need to poke libusb to experience the hang.  Beginning to suspect something nastier...

Other info:
- libusb-1.0, version 1.0.2
- libuvc, version 0.0.5 (09f1822)

```
#include <libuvc/libuvc.h>
#include <unistd.h>

void cb(struct uvc_frame* /*frame*/, void* /*obj*/)
{
  printf("[RECV] Frame\n");
}

class uvc_wrapper_t
{
public:
  uvc_context_t       *ctx;
  uvc_device_t        *dev;
  uvc_device_handle_t *devh;
  uvc_stream_ctrl_t   ctrl;
  uvc_error_t         err;
  bool                streaming;

  uvc_wrapper_t(const char* id) : ctx(NULL), dev(NULL), devh(NULL), streaming(false)
  {
    err = uvc_init(&ctx, NULL);
    if (err < 0) return;
    err = uvc_find_device(ctx, &dev, 0, 0, id);
    if (err < 0) return;
    err = uvc_open(dev, &devh);
    if (err < 0) return;
    err = uvc_get_stream_ctrl_format_size(devh, &ctrl, UVC_FRAME_FORMAT_UNCOMPRESSED, 1280, 960, 0);
    if (err < 0) return;
    err = uvc_start_streaming(devh, &ctrl, cb, NULL, 0);
    if (err < 0) return;
    streaming = true;
  }
  ~uvc_wrapper_t()
  {
    if (streaming) uvc_stop_streaming(devh);
    if (devh) uvc_close(devh);
    if (ctx) uvc_exit(ctx);
  }
};

int main(int argc, char** argv)
{                                       // Test overview:
  uvc_wrapper_t wrapper(argv[1]);       // 1. Begin streaming
  if (wrapper.err < 0)
  {
    uvc_perror(wrapper.err, "Error");
    return 1;
  }

  sleep(1);                             // 2. Sleep for a hot second

  printf("Process complete.\n");

  return 0;                             // 3. That's it!
}
```

nussetorten, 2017-11-17:
As I wrote the previous comment, I noticed how old a version of libuvc I was using.

Updating to 0.0.6 resolved my issues.

Aren't bugs which resolve themselves the best?

closed 2017-11-17 by nussetorten

openastroproject, 2017-11-17:
I think this problem could be the one fixed by my change here: https://github.com/ktossell/libuvc/commit/c3ec064d135b598273e2fae6679afc5f9ff8db39 in which case moving to the later version is definitely the correct thing to do.

nussetorten, 2017-11-20:
@openastroproject thanks for the extra digging, this looks to be the culprit 👍

PeterPeterPan, 2020-09-30:
@openastroproject  thank you so so so so so much.