libuvc/libuvc #92: CentOS7 libusb-1.0 not found

aFunnyBit, 2017-12-14:
```
git clone https://github.com/ktossell/libuvc
cd libuvc
mkdir build
cd build
cmake ..
```

ERROR does not complete with the error libusb-1.0 not found

```
[tdev@localhost build]$ cmake ..
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.27.1") 
-- checking for module 'libusb-1.0'
--   libusb-1.0 not found
-- Building libuvc with JPEG support.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tdev/libuvc/build
```

Is there anyway I can manually point to the libusb I have installed

mcuee, 2018-07-09:
Why not just install the libusb package under CentOS?

waddlesplash, 2019-07-06:
Indeed, why not?

jwlodek, 2019-08-23:
On CentOS7, libusb-1.0 needs the libusbx-devel yum package I believe
```
yum install libusbx-devel
```
and you will obviously need the regular libusb package as well
```
yum install libusb-devel
```