libuvc/libuvc #93: Get invaild mode error when using uvc_get_stream_ctrl_format_size

andymatos, 2018-01-07:
I have a uvc camera and try to do some image processing work on its stream. I am working under ubuntu 16.04. All works fine if I use opencv to open the device. Since I'd like to open the device through the vid and pid, I try to use libuvc to get the video stream.
So I tried the test.c in libuvc and get the following output:

Device opened
[device.c:871/uvc_claim_if] claiming interface 0
[device.c:880] end uvc_claim_if (0)
[device.c:326/uvc_open] libusb_submit_transfer() = 0
[device.c:343] end uvc_open (0)
[device.c:481] begin uvc_get_device_descriptor
[device.c:523] end uvc_get_device_descriptor (0)
DEVICE CONFIGURATION (1234:0101/[none]) ---
[device.c:535] begin uvc_free_device_descriptor
[device.c:548] end uvc_free_device_descriptor
Status: idle
VideoControl:
	bcdUVC: 0x0100
VideoStreaming(1):
	bEndpointAddress: 0
	Formats:
END DEVICE CONFIGURATION
bmHint: 0000
bFormatIndex: 0
bFrameIndex: 0
dwFrameInterval: 0
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 4720
dwMaxVideoFrameSize: 0
dwMaxPayloadTransferSize: 4198000
bInterfaceNumber: 253
get_mode: Invalid mode (-51)

In test.c I only change the vid and pid in uvc_find_device and the width and height of the frame and fps in uvc_get_stream_ctrl_format_size. My device support 1280*960@15fps and 1280*720@10fps. I have tried both resolutions and get the same invaild mode error.

Am I missing something important?

maozhenyu00, 2018-01-09:
hi,I also met this problem.
The get_mode only know  "UVC_FRAME_FORMAT_MJPEG"

yoiang, 2018-06-07:
I'm not sure if it helps but I had a conflicting version of libuvc in another library (librealsense) that had expectations that resulted in an invalid mode error.

BoysFight, 2021-09-18:
hi,I also hav this problem.
Mark.