libuvc/libuvc #95: uvc_mjpeg2yuyv return -98

Shuyun6, 2018-01-25:
In frame-mjpeg.c, uvc_mjpeg2yuyv return -98, why???

waddlesplash, 2019-07-06:
I don't really understand what you mean; could you elaborate?

alterli, 2019-11-25:
I have the same err:

01-01 10:21:35.279: E/libUVCCamera(1633): [1674*UVCPreview.cpp:423:addPreviewFrame]:enter...
01-01 10:21:35.279: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=64
01-01 10:21:35.279: W/libUVCCamera(1633): [1673*frame-mjpeg.c:481:uvc_mjpeg2yuyv]:err=19537 7140 640 480 
01-01 10:21:35.280: W/libUVCCamera(1633): [1673*diag.c:88:uvc_perror]:failed uvc_mjpeg2yuyv:Unknown error (-98)
01-01 10:21:35.280: E/libUVCCamera(1633): [1673*UVCPreview.cpp:438:waitPreviewFrame]:enter...
01-01 10:21:35.280: W/libUVCCamera(1633): [1673*frame-mjpeg.c:481:uvc_mjpeg2yuyv]:err=19537 5648 640 480 
01-01 10:21:35.280: W/libUVCCamera(1633): [1673*diag.c:88:uvc_perror]:failed uvc_mjpeg2yuyv:Unknown error (-98)

01-01 10:21:35.280: E/libUVCCamera(1633): [1673*UVCPreview.cpp:438:waitPreviewFrame]:enter...
01-01 10:21:35.283: E/libUVCCamera(1633): [1674*UVCPreview.cpp:423:addPreviewFrame]:enter...
01-01 10:21:35.283: W/libUVCCamera(1633): [1673*frame-mjpeg.c:481:uvc_mjpeg2yuyv]:err=19537 564 640 480 
01-01 10:21:35.284: W/libUVCCamera(1633): [1673*diag.c:88:uvc_perror]:failed uvc_mjpeg2yuyv:Unknown error (-98)
01-01 10:21:35.284: E/libUVCCamera(1633): [1673*UVCPreview.cpp:438:waitPreviewFrame]:enter...
01-01 10:21:35.307: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=16
01-01 10:21:35.307: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=32
01-01 10:21:35.307: I/libuvc/stream(1633): [stream.c:868:_uvc_process_payload_iso]:bad packet:status=0xca
01-01 10:21:35.308: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=96
01-01 10:21:35.311: I/libuvc/stream(1633): [stream.c:868:_uvc_process_payload_iso]:bad packet:status=0xe2
01-01 10:21:35.315: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=120
01-01 10:21:35.319: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=128
01-01 10:21:35.319: I/libuvc/stream(1633): [stream.c:868:_uvc_process_payload_iso]:bad packet:status=0x40
01-01 10:21:35.320: E/libUVCCamera(1633): [1674*UVCPreview.cpp:423:addPreviewFrame]:enter...
01-01 10:21:35.320: W/libUVCCamera(1633): [1673*frame-mjpeg.c:481:uvc_mjpeg2yuyv]:err=19537 32 640 480 
01-01 10:21:35.320: W/libUVCCamera(1633): [1673*diag.c:88:uvc_perror]:failed uvc_mjpeg2yuyv:Unknown error (-98)
01-01 10:21:35.320: E/libUVCCamera(1633): [1673*UVCPreview.cpp:438:waitPreviewFrame]:enter...
01-01 10:21:35.323: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=216
01-01 10:21:35.323: E/libUVCCamera(1633): [1674*UVCPreview.cpp:423:addPreviewFrame]:enter...
01-01 10:21:35.324: W/libUVCCamera(1633): [1673*frame-mjpeg.c:481:uvc_mjpeg2yuyv]:err=19537 12 640 480 
01-01 10:21:35.324: W/libUVCCamera(1633): [1673*diag.c:88:uvc_perror]:failed uvc_mjpeg2yuyv:Unknown error (-98)
01-01 10:21:35.324: E/libUVCCamera(1633): [1673*UVCPreview.cpp:438:waitPreviewFrame]:enter...
01-01 10:21:35.347: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=16
01-01 10:21:35.347: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=32
01-01 10:21:35.347: I/libuvc/stream(1633): [stream.c:868:_uvc_process_payload_iso]:bad packet:status=0xca
01-01 10:21:35.348: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=96
01-01 10:21:35.351: I/libuvc/stream(1633): [stream.c:868:_uvc_process_payload_iso]:bad packet:status=0xe2
01-01 10:21:35.351: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=15
01-01 10:21:35.355: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=120
01-01 10:21:35.359: I/libuvc/stream(1633): [stream.c:922:_uvc_process_payload_iso]:bogus packet: actual_len=12, header_len=128