libuvc/libuvc #98: Unable to run example.c. Undefined reference to uvc_init and to rest.

prnvjb, 2018-04-28:
#include "libuvc/libuvc.h"
#include <stdio.h>

int main(int argc, char **argv) {
  uvc_context_t *ctx;
  uvc_device_t *dev;
  uvc_device_handle_t *devh;
  uvc_stream_ctrl_t ctrl;
  uvc_error_t res;

  res = uvc_init(&ctx, NULL);

  return 0;
}

ERROR:: undefined reference to "uvc_init"
it is also showing this error to  "uvc_find_device" 

I installed everything correctly. I'm using ubuntu 16. Did I miss anything?
Please guide me. I'm a newbie.

Thnaks.

drjasonharrison-vp-eio, 2018-10-16:
What is your compilation command? You may need to provide the paths to the include and library files.
Please include more output from the compiler.

mcuee, 2026-09-06:
Close this old issue for now.

closed 2026-09-06 by mcuee