libuvc/libuvc #100: Not easy to build on Windows with Visual Studio 2015

logieestrada, 2018-06-29:
Im starting to play around with libuvc, but unable to successfully compile
 - I added pthreads_win32
- Now it is looking for libgen.h

Are there instructions how to compile on Windows?

mcuee, 2018-07-09:
I used MSYS2 and it seems to be not difficult. I have not tried Visual Studio myself. Anyway, you may want to try the fork here and it seems to offer better Windows support, including instructions to build under Windows.
https://github.com/pupil-labs/libuvc
https://github.com/pupil-labs/libuvc/blob/master/INSTALL_WINDOWS.md (the first part is not necessary, you just need to use latest libusb-1.0.22 which support isoc transfer using libusbk).

sayezz, 2020-05-08:
Wanna just thank you. Your info helped me to get it working under Win10, Visual Studio 2017

waddlesplash, 2020-05-08:
Yes, I asked at https://github.com/pupil-labs/libuvc/issues/34 if they could upstream their changes, but there has been no reply, so it may need to be done manually.

closed 2020-05-08 by waddlesplash