libuvc/libuvc #105: Build failed due to "sys/time.h" is not available on windows

attina, 2018-09-27:
Following is the fail log.

2>c:\projects\lightsaber2\common\libuvc\include\libuvc\libuvc.h(10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
2>ctrl-gen.c
2>c:\projects\lightsaber2\common\libuvc\include\libuvc\libuvc.h(10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
2>device.c
2>c:\projects\lightsaber2\common\libuvc\include\libuvc\libuvc.h(10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
2>diag.c
2>c:\projects\lightsaber2\common\libuvc\include\libuvc\libuvc.h(10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
2>frame.c
2>c:\projects\lightsaber2\common\libuvc\include\libuvc\libuvc.h(10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
2>init.c
2>c:\projects\lightsaber2\common\libuvc\include\libuvc\libuvc.h(10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
2>stream.c
2>c:\projects\lightsaber2\common\libuvc\include\libuvc\libuvc.h(10): fatal error C1083: Cannot open include file: 'sys/time.h': No such file or directory
2>Generating Code...

kinsou, 2018-11-30:
#include <sys/time.h> to  #include <time.h>

wangjia184, 2024-09-08:
then failed with another error : 

> libuvc_internal.h(12): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory

Does it work in MSVC toolchain?