libuvc/libuvc #107: need example for use of uvc_stream_get_frame

drjasonharrison-vp-eio, 2018-10-03:
I'm trying to figure out how to use uvc_stream_get_frame:
```
uvc_error_t uvc_stream_get_frame	(	uvc_stream_handle_t * 	strmh,
    uvc_frame_t ** 	frame,
    int32_t 	timeout_us 
)		
```
Through examination of the internals of libuvc and testing I've determined:
* uvc_start_streaming must be called with a NULL for the callback parameters
* uvc_stream_open_ctrl appears to be the function used to get the uvc_stream_handle_t for the call to uvc_stream_get_frame

So my current code is:
```
res = uvc_find_device(ctx, &dev, 0, 0, NULL); 
res = uvc_open(dev, &devh);
res = uvc_get_stream_ctrl_format_size(devh, &ctrl, UVC_FRAME_FORMAT_GRAY16, 320, 256, 60); 
res = uvc_stream_open_ctrl(devh, &strmhp, &ctrl);
res = uvc_start_streaming(devh, &ctrl, NULL, NULL, 0);
res = uvc_stream_get_frame(strmhp, &frame, 0);
```
However, uvc_strart_streaming  returns `Busy (-6)`

I get the same error if I call uvc_start_streaming first, but then uvc_stream_open_ctrl  returns `Busy (-6)`

grzegb, 2018-11-28:
Call:
```
res = uvc_stream_open_ctrl(devh, &strmhp, &ctrl);
res = uvc_stream_start(strmhp, NULL, NULL, 0);
res = uvc_stream_get_frame(strmhp, &frame, 0);
```

jumpy88, 2020-03-07:
Hello @drjasonharrison-vp-eio have you solved it? Did you try the solution proposed by @grzegb?

MarsXiaolei, 2024-06-27:
res = uvc_stream_open_ctrl(devh, &strmhp, &ctrl);
res = uvc_stream_start(strmhp, NULL, NULL, 0);
res = uvc_stream_get_frame(strmhp, &frame, 0);
Thanks @grzegb , I have tested this code and can get the frame data. First call the function uvc_get_stream_ctrl_format_size to get the ctrl.

closed 2026-09-06 by mcuee

drjasonharrison, 2026-09-07:
Sorry for not getting back to you sooner, I was laid off in February of 2020 and lost access to the account and the pandemic wiped my immediate focus. Glad to see the explanation!