libuvc/libuvc #108: How to reset camera when "attempt to claim already-claimed interface" is reported

drjasonharrison-vp-eio, 2018-10-16:
When the error "attempt to claim already-claimed interface" is reported is there any known way to reset the camera? libuvc appears to work ok but I'm worried that this error will cause problems later on.

waddlesplash, 2019-07-06:
I think the error means that the USB device was already "claimed" by some other program or driver.

openastroproject, 2019-07-08:
This is code I added some time back to flag up a probable bug that I've not been able to track down yet.  It means  the current process is making attempts to claim an interface that it has already claimed.

I don't think any reset should be required, as the function just becomes a no-op if it detects that the interface is already claimed.

My recollection is that Linux doesn't care if you attempt to claim an interface more than once, but it does cause problems on (some versions of?) MacOS.