libuvc/libuvc #109: Issue with finding JPEG library when compiling.

RustyRaptor, 2018-11-02:
I am trying to compile libuvc with JPEG support for GetThermal. When I run cmake I get this 

```
CMake Warning at CMakeLists.txt:58 (message):
  JPEG not found.  libuvc will not support JPEG decoding.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/ziad/Documents/src/ETV/WIND/GetThermal/libuvc/build
```

I am running Ubuntu 18.04 and I tried installing every libjpeg development package there is. Including. 
libjpeg-dev
libjpeg9-dev
libjpeg8-dev
libjpeg-turbo8-dev
libjpeg62-dev

Is there some other jpeg library or a different version I need? would be nice to have a list of dependencies that's easy to reach somewhere.

Domihk, 2019-01-24:
Hi, have you manage to solve this issue? I find myself in a similar situation.

tarhan, 2019-07-15:
In Ubuntu 18.04 it called: `libjpeg-turbo8-dev`

waddlesplash, 2019-07-24:
Please recheck after #125.

RustyRaptor, 2019-07-26:
As you might have guessed I am no longer working on this. The project I am working on is on hold. If I come back to it at a later time I'll let you all know.

RustyRaptor, 2019-07-26:
Actually I'll just compile it right now real quick. 

and Great Success. Here is the output. Have a nice day. 

```
 ~  git clone https://github.com/ktossell/libuvc
cd libuvc
mkdir build
cd build
cmake ..
make && sudo make install
Alias tip: g clone https://github.com/ktossell/libuvc
cd libuvc
mkdir build
cd build
cmake ..
make && sudo make install
Cloning into 'libuvc'...
remote: Enumerating objects: 57, done.
remote: Counting objects: 100% (57/57), done.
remote: Compressing objects: 100% (47/47), done.
remote: Total 841 (delta 25), reused 31 (delta 10), pack-reused 784
Receiving objects: 100% (841/841), 457.98 KiB | 2.09 MiB/s, done.
Resolving deltas: 100% (488/488), done.
-- The C compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- No build type selected, default to Release
-- No target type selected, default to both shared and static library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.21
-- libusb-1.0 found using pkgconfig
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so (found version "80") 
-- Found JPEG library using standard module
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Building libuvc with JPEG support.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ziad/libuvc/build
Scanning dependencies of target uvc
[  5%] Building C object CMakeFiles/uvc.dir/src/ctrl.c.o
[ 10%] Building C object CMakeFiles/uvc.dir/src/ctrl-gen.c.o
[ 15%] Building C object CMakeFiles/uvc.dir/src/device.c.o
[ 20%] Building C object CMakeFiles/uvc.dir/src/diag.c.o
[ 25%] Building C object CMakeFiles/uvc.dir/src/frame.c.o
[ 30%] Building C object CMakeFiles/uvc.dir/src/init.c.o
[ 35%] Building C object CMakeFiles/uvc.dir/src/stream.c.o
[ 40%] Building C object CMakeFiles/uvc.dir/src/misc.c.o
[ 45%] Building C object CMakeFiles/uvc.dir/src/frame-mjpeg.c.o
[ 50%] Linking C shared library libuvc.so
[ 50%] Built target uvc
Scanning dependencies of target uvc_static
[ 55%] Building C object CMakeFiles/uvc_static.dir/src/ctrl.c.o
[ 60%] Building C object CMakeFiles/uvc_static.dir/src/ctrl-gen.c.o
[ 65%] Building C object CMakeFiles/uvc_static.dir/src/device.c.o
[ 70%] Building C object CMakeFiles/uvc_static.dir/src/diag.c.o
[ 75%] Building C object CMakeFiles/uvc_static.dir/src/frame.c.o
[ 80%] Building C object CMakeFiles/uvc_static.dir/src/init.c.o
[ 85%] Building C object CMakeFiles/uvc_static.dir/src/stream.c.o
[ 90%] Building C object CMakeFiles/uvc_static.dir/src/misc.c.o
[ 95%] Building C object CMakeFiles/uvc_static.dir/src/frame-mjpeg.c.o
[100%] Linking C static library libuvc.a
[100%] Built target uvc_static
[sudo] password for ziad: 
[ 50%] Built target uvc
[100%] Built target uvc_static
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/libuvc.so.0.0.6
-- Installing: /usr/local/lib/libuvc.so.0
-- Installing: /usr/local/lib/libuvc.so
-- Installing: /usr/local/include/libuvc/libuvc.h
-- Installing: /usr/local/include/libuvc/libuvc_config.h
-- Installing: /usr/local/lib/libuvc.a
-- Up-to-date: /usr/local/include/libuvc/libuvc.h
-- Up-to-date: /usr/local/include/libuvc/libuvc_config.h
-- Installing: /usr/local/lib/cmake/libuvc/libuvcTargets.cmake
-- Installing: /usr/local/lib/cmake/libuvc/libuvcTargets-release.cmake
-- Installing: /usr/local/lib/cmake/libuvc/FindLibUSB.cmake
-- Installing: /usr/local/lib/cmake/libuvc/FindJpegPkg.cmake
-- Installing: /usr/local/lib/cmake/libuvc/libuvcConfigVersion.cmake
-- Installing: /usr/local/lib/pkgconfig/libuvc.pc
-- Installing: /usr/local/lib/cmake/libuvc/libuvcConfig.cmake

```

RustyRaptor, 2019-07-26:
Also successfully built GetThermal so that works too.

closed 2019-07-26 by RustyRaptor

tarhan, 2019-07-27:
Glad to hear.