libuvc/libuvc #113: libuvc test code does not work on ELP 8M USB camera

ArtlyStyles, 2019-02-04:
I bought two USB cameras from Amazon, one has 2M and  the other one has 8M sensor. I tested them with libuvc on Nvidia Jetson Tx2 running Linux. I download the libuvc code and compile them successfully. 

When I used this camera https://www.amazon.com/gp/product/B00LQ854AG, the test program "uvc_test" and "example" worked perfectly. 

However, when I used this camera https://www.amazon.com/gp/product/B00LQ854AG, which has a 8M sensor, I could NOT run either of the test programs. The print-out is attached below.

In the code, I was requesting 640x480 30 frame YUYV, which is supported in both cameras. The two cameras come from the same manufacturer, just different sensors. Also I could use both cameras successfully on Windows with the "Camera" app, so I guess the program is from the libuvc code. 

Anyone knows how to fix this so I can use the 8M camera with libuvc?

uvc_test print-out when I ran the 8M camera:

UVC initialized
Device found
Device opened
DEVICE CONFIGURATION (05a3:8830/SN0179) ---
Status: idle
VideoControl:
	bcdUVC: 0x0100
VideoStreaming(1):
	bEndpointAddress: 0
	Formats:
END DEVICE CONFIGURATION
Print diag done
bmHint: 1978
bFormatIndex: 64
bFrameIndex: 0
dwFrameInterval: 0
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 960
dwMaxVideoFrameSize: 127
dwMaxPayloadTransferSize: 2648451176
bInterfaceNumber: 0
get_mode: Invalid mode (-51)
Device closed
UVC exited

ArtlyStyles, 2019-02-07:
Anyone know why?

cpicanco, 2019-02-09:
I am afraid I cannot help you. If of any interest, I have a working uvc-compatible ELP camera that has libuvc as a dependency (for this project here https://github.com/pupil-labs/pyuvc/issues/10).

attackordie, 2020-05-27:
Has there been any progress on this issue?

deeprun, 2021-09-03:
Looks like current libUVC does not support his mode. I used opencv, and it worked. But libUVC does not work.

deeprun, 2021-09-03:
@waddlesplash can you please implement libUVC support for this camera?

waddlesplash, 2021-09-03:
Ultimately I am just one of the "maintainers" of libuvc, and I have done comparatively little development. I actually don't have permission to merge my own PRs, and so I've had one sitting around myself for some time (#144).

deeprun, 2021-09-03:
hi @waddlesplash If you know how to fix this, could you show the code here?