libuvc/libuvc #114: Does setting the exposure time of UVC cameras through libuvc change framerate?

jwlodek, 2019-02-27:
I am developing a driver for certain scientific software using libuvc, and it would be very useful to be able to achieve variable frame rates, and not just the ones described when getting camera information. Currently, creating a stream with a framerate not listed throws a UVC Error, but would changing the exposure time affect the framerate?

dmikushin, 2020-05-26:
The purpose of variable frame rate alone is unclear. Nonetheless, the easiest variation could be achieved by means of frame skipping with different multipliers. Yes, exposure starts to affect the frame rate, if set large enough. But you won't be happy with such a way of variable fps, as your images would gather too much light (overexposure).

dmikushin, 2020-05-26:
In fact, fps rate of camera could be set to almost arbitrary value by varying camera clock frequency. Raspberry Pi community has nice examples of camera drivers made to achieve unrealistically high FPS (obviously, with strong artefacts). But frequency is a driver-level setting and clearly is beyond the scope of libuvc.