libuvc/libuvc #118: Set Auto Exposure error

ArtlyStyles, 2019-06-21:
When I set the auto exposure uing the code below
`uvc_set_ae_mode(_devh, 2);`
I got a "Pipe (-9)" error. What does this mean?

I am using this camera https://www.amazon.com/gp/product/B00LQ854AG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

dmikushin, 2020-03-09:
Same for me on Ubuntu with builtin laptop camera.

ktossell, 2020-03-10:
Is it possible that the camera doesn't support the full-auto AE mode? I get the same error when testing with my laptop's builtin camera...

I added this to the uvc_set_ae_mode docs back in 2014:
> Most cameras provide manual mode and aperture priority mode.

so I probably thought at the time that the other two modes (shutter priority, and fully automatic) were not commonly supported. @dmikushin Does AE work with your builtin camera if you use `UVC_AUTO_EXPOSURE_MODE_APERTURE_PRIORITY (8)`?

I've updated https://github.com/libuvc/libuvc/pull/138 with some code that will first try AUTO mode and then fall back to APERTURE_PRIORITY mode if AUTO didn't work.

dmikushin, 2020-03-11:
**EXCELLENT.** 8 sets Linux camera brightness to the expected usual automatic behavior. Many thanks for your kind help!

dmikushin, 2020-03-11:
Now, interestingly when I test (8) on Mac, it now gets "Pipe error" there. So, in order to make both Linux and MacOS happy I ended up with the following:

```
#ifdef __APPLE__
                const uint8_t UVC_AUTO_EXPOSURE_MODE_AUTO = 2;
                res = uvc_set_ae_mode(devh, UVC_AUTO_EXPOSURE_MODE_AUTO);
#else
                const uint8_t UVC_AUTO_EXPOSURE_MODE_APERTURE_PRIORITY = 8;
                res = uvc_set_ae_mode(devh, UVC_AUTO_EXPOSURE_MODE_APERTURE_PRIORITY);
#endif
```