libuvc/libuvc #124: About  Libuvc 's  STILL Image capture

yuliang8, 2019-07-11:
Hi all:
    The latest version supports still image features. But We found that It Only parse frame description after Initial, cannot to Capture still Image when host send trigger command to device. so we want to know, whether any one give some help about STILL Image function

 Thanks.

waddlesplash, 2019-07-12:
I'm not quite sure what this is asking...

(Still image support was implemented by @brunomarchand fwiw.)

brunomarchand, 2019-07-12:
Hello,
Although I did not commit an example how to use the API, here is one from the pull request thread: https://github.com/libuvc/libuvc/pull/89#issuecomment-348552623

I hope this will help.

yuliang8, 2019-07-12:
> Although I did not commit an example how to use the API, here is one from the pull request thread: #89 (comment)

Thank you very much 

发自我的 iPhone

closed 2019-07-24 by waddlesplash