libuvc/libuvc #126: Different result when calling uvc_get_device_list - Linux vs Android

wlbe4, 2019-07-25:
Hi, 
I have compiled the libuvc on both Linux PC and a rooted Android phone (Dev phone running Android 9.0).
The test app on the phone is executed on a root shell terminal.

I have connected a Logitec web cam and only wanted to detect it (currently no openCV code is used to draw the video).

When running the test on Linux, I am getting 2 interfaces Logitec web cam.
The first one has 1 altsettings (class 14 and subclass 1) - This is the SC_VIDEOCONTROL.
The second one has 12 altsettings (class 14 and subclass 2) - This is the SC_VIDEOSTREAMING.

When running the same test on the Android device, I am getting only 1 interface from the webcam (the SC_VIDEOCONTROL).

(I have modified the device.c file so it will print out the number of interfaces I get from each device and the num of altsettings)

Any advice would be highly appreciated.

waddlesplash, 2019-07-27:
What libusb for Android are you using? It's possible that is the main difference here.

wlbe4, 2019-07-28:
Thanks waddlesplash,
I have been using libusb version: 1.0.22.11340 (used it in few Android projects for the past year mainly reading bulk data from an FX3 development kit device).
Today I took latest version (1.0.23.11377) and things got worse.
Now it doesn't list any device.
I will continue to investigate and debug.
If you have any idea what may cause the difference, please advice.

wlbe4, 2019-07-28:
Found the difference between the latest version and the one I was using previously.
I made a change in libusb sources to allow running in Android shell (made it last year and forgot about it...).
The problem is that the define __ANDROID__ is set when we compile the library via ndk-build and usually the code that is under this define assumes that the library is used in an Android application.
In my case, even though this is an android platform, when I run a test application from shell it is more like a Linux rather than Android,
The change I made was to the file linux_usbfs.c, function linux_scan_devices:
static int linux_scan_devices(struct libusb_context *ctx)
{
	int ret = 0;

	usbi_mutex_static_lock(&linux_hotplug_lock);

#if defined(USE_UDEV)
	ret = linux_udev_scan_devices(ctx);
#else //if !defined(__ANDROID__)
	// When running on Android rooted device from shell application we need to be able to scan devices
	ret = linux_default_scan_devices(ctx);
#endif

	usbi_mutex_static_unlock(&linux_hotplug_lock);

	return ret;
}

So now, after applying this change to the latest libusb, I am back at the point where I was last week.
In other words, the Logitec web cam only lists the VIDEOCONTROL but not the VIDEOSTREAMING.

I will continue to investigate and debug...

wlbe4, 2019-07-28:
After a long day of debugging and comparing traces, I found that libusb compiled with clang introduce a bug in the descriptor parsing. Adding a "printf" call in a certain point in the code (in descriptor.c line 281 after the call to usbi_parse_descriptor(buffer, "bb", &header, 0);) solved the problem. 
I am still not sure what is the real issue with clang but when I use an older ndk (r15c) which allows me to use gcc, there is no issue at all.
If I have the time I will continue to investigate to find what caused the clang to generate the faulty code.
In the meantime, I will use the gcc generated lib of libusb which works fine.

waddlesplash, 2019-07-28:
Well, please report that bug to libusb then; likely it's some badly optimized code or the like.

closed 2019-07-28 by waddlesplash