libuvc/libuvc #128: Request for function that can get data type / color mode

jwlodek, 2019-08-12:
In my application that uses libuvc, I need information regarding the frame contents in terms of mono/rgb and 8bit/16bit prior to starting streaming. I used libuvc to write a generic scientific driver for UVC cameras meaning that the same code runs for each camera. 

I would like to make a function that can only allow users to select available color mode / bit depth combinations, however, I can't seem to find a way of doing this easily internally in libuvc. This is only really a problem when the camera sends `UVC_FRAME_FORMAT_UNCOMPRESSED`, when sometimes it is rgb and sometimes grayscale without a way of easily knowing as far as I can tell.

I was thinking about developing something like this myself but I figured I would ask in case anyone has already found/developed a solution.

openastroproject, 2019-08-12:
Is checking "fourccFormat" in the format descriptions insufficient to achieve what you need?

jwlodek, 2019-08-12:
Unfortunately no. The fourcc format tells me what format the frame will be, but not necessarily whether it will be 8 bit or 16 bit and if it will be color or grayscale. Note that this is only an issue with the `UNCOMPRESSED` format, as it seems to have instances of being both 8 bit rgb, 8 bit grayscale, and 16 bit rgb. (I have cameras that fall into all 3 categories).

All of the other formats are locked into a particular color mode/bit depth and thus do not have this problem

openastroproject, 2019-08-12:
What values do you get for the four cc value in those cases?