libuvc/libuvc #132: Auto exposure is not working

deeprun, 2019-10-10:
Seems auto exposure is not working. Anyone tried that?

dmikushin, 2020-03-09:
Yes, same for me (very dark). It's actually even more interesting. I've opened MacOS' builtin Photo Booth before and after launching camera from within libuvc. First time Photo Booth video was bright "as usual". But when I opened Photo Booth *after* testing libuvc, video became a lot darker. And there are no settings to reset it back. Congratulations @ktossell hijacking camera so brutally, that even MacOS can't stand it :+1:

dmikushin, 2020-03-09:
Sorry, actually auto exposure works. @deeprun please make sure you supply value of 2:

```
const uint8_t UVC_AUTO_EXPOSURE_MODE_AUTO = 2;
res = uvc_set_ae_mode(devh, UVC_AUTO_EXPOSURE_MODE_AUTO);
```

ktossell, 2020-03-10:
I've posted a PR that applies this correction to the example code; does this look all right? https://github.com/libuvc/libuvc/pull/138

dmikushin, 2020-03-10:
@ktossell thank you for looking. Right. Yet, as mentioned in #118 auto exposure won't get through in Linux. Still to be investigated on that side.