libuvc/libuvc #134: Error while closing uvc stream handle after getting UVC_ERROR_CALLBACK_EXISTS from uvc_stream_get_frame

chirag26495, 2019-11-19:
**Working Environment \:**
Using Debian Linux,
Using Camera:
$ lsusb
Bus 001 Device 013: ID 04b4:00f8 Cypress Semiconductor Corp. 
Bus 001 Device 002: ID 0424:2512 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Using libusb version v1.0.22,
Using libuvc version v0.0.5,

**After Execution of Code:**
uvc_frame_t *frame_ptr=NULL;
res = uvc_stream_get_frame(strmh, &frame_ptr, -1);
// res value returned here = **-52** , after this camera error, the camera object is being destroyed in which following commands are executed in order:
cout<<"UVC : "<<current_datetime_to_string()<<" : Attempting disconnection"<<std::endl;
cout<<"UVC : free bgr frame"<<std::endl;
uvc_free_frame(bgr);
cout<<"UVC : closing stream"<<std::endl;
uvc_stream_close(strmh);
// throws error:

**Error Traceback info**
UVC : 2019_11_19_18_34_18 : frame timeout : **-52**
CAM CTRL : 2019_11_19_18_34_18 : camera error, disconnecting
UVC : 2019_11_19_18_34_18 : Attempting disconnection
UVC : free bgr frame
UVC : closing stream

Thread 1 "<****>" received signal SIGSEGV, Segmentation fault.
0x0000ffffbf67bd34 in pthread_mutex_lock () from /lib/aarch64-linux-gnu/libpthread.so.0
(gdb) bt
#0  0x0000ffffbf67bd34 in pthread_mutex_lock () from /lib/aarch64-linux-gnu/libpthread.so.0
#1  0x0000ffffbe3b0a10 in libusb_cancel_transfer () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#2  0x0000ffffbe425c0c in uvc_stream_stop () from /home/linaro/development/build/libCapture/libuvc/libuvc.so
#3  0x0000ffffbe425d5c in uvc_stream_close () from /home/linaro/development/build/libCapture/libuvc/libuvc.so
#4  0x0000aaaaaab6ee40 in CamUVC::attempt_disconnection() ()
#5  0x0000aaaaaab6f0a0 in CamUVC::~CamUVC() ()

If any details regarding above are missing kindly ping, 
Does anybody have a clue here? :X

chirag26495, 2019-11-20:
Solved! It was actually before the UVC_CALLBACK_ERROR_EXISTS, the uvc stream handle wasn't closed and opened properly.

closed 2019-11-20 by chirag26495

pandeypradyumn, 2022-06-30:
@chirag26495 can you tell me the solution for this ?