libuvc/libuvc #136: Issues compiling on Windows

avsdev-cw, 2020-02-26:
For `a cross-platform library for USB video devices` there are quite a few issues I've encountered trying to get libuvc on Windows. So far they include the list below, I'm sure more may be added and I welcome any proper fixes (as apposed to my scratchy work-arounds).

- Depends on libJpeg (whilst optional, in reality most devices stream mjpeg, so making it optional isn't much use) {I got around this by using gnuwin32's libjpeg installer}
- cmake files do not have any definitions on what to do on Windows (package files FindJpegPkg FindLibUSB). Manually adding JPEG_FOUND, JPEG_INCLUDE_DIRS, JPEG_LIBRARIES and the respective LibUSB_* declarations gets you through configuration, but fails generation with:
`Target "uvc" links to target "LibUSB::LibUSB" but the target was not found.`
- libusb.h header file includes <sys/time.h>. Using <time.h> does not work because:
`error C2079: 'capture_time' uses undefined struct 'timeval'`
- Silently depends on pthread {I downloaded pthread-win32 but because of the silent include, there doesn't seem to be any configuration settings in cmake that are used. Manually running cl.exe is blocked by the time.h issue above}

avsdev-cw, 2020-02-26:
Example modification to the cmake Find* files (which assumes *_FOUND, *_INCLUDE_DIR AND *_LIBRARIES are defined):
```

if (MSVC OR MINGW)
  if(LibUSB_FOUND)
    if (NOT TARGET LibUSB::LibUSB)
      add_library(LibUSB::LibUSB UNKNOWN IMPORTED)
      if(LibUSB_INCLUDE_DIR)
	    set_target_properties(LibUSB::LibUSB PROPERTIES
	      INTERFACE_INCLUDE_DIRECTORIES ${LibUSB_INCLUDE_DIR}
	    )
      endif()
      if(LibUSB_LIBRARIES)
        set_target_properties(LibUSB::LibUSB PROPERTIES
	      IMPORTED_LINK_INTERFACE_LANGUAGES "C" 
	      IMPORTED_LOCATION ${LibUSB_LIBRARIES}
        )
      endif()
    endif()
  endif()
  return()
endif()
```

dmikushin, 2020-03-11:
I'd recommend to bundle libusb and jpeglib as CMake submodules - that would be the smartest and the most portable way.

Be careful with `<chrono>` clocks, their precision (resolution) appears to be lower than expected, many people already stumbled upon this.

avsdev-cw, 2020-03-11:
After some further digging I did spot a pr #68 which addresses a number of the issues, however it does drop jpeg support and I ended up pulling the pupil-labs/libuvc master and cherry picking some changes from both to get a working build. It wasn't pretty!

Given the age and state of the PR, perhaps it would be worth starting a new one that can be merged in faster?

nospam2k, 2020-07-03:
962557346 - Can you please outline how you compiled on Windows. I'm really stuck. I have no problem doing this on Mac/Linux but I'm lost on Windows.

962557346, 2020-07-04:
@nospam2k Hello, I understand your feelings. Firstly, I will outline all the steps.
1) I downloaded the modified version ([libuvc-pupil-master](https://github.com/pupil-labs/libuvc)).
2) I downloaded the [libusb](https://sourceforge.net/projects/libusb/) and [pthreadVC2](https://www.mirrorservice.org/sites/sourceware.org/pub/pthreads-win32/).  Both two packages have been complied. Remember  the pthreads-w32-2-9-1-release.zip at the bottom of this page.
3) I modified the CMakeLists.txt, the changes are as follows:
```
if(WIN32)
  find_library(LIBUSB_LIBRARIES libusb-1.0
  PATHS E:/3.Projects/Eyes/packages/libuvc-pupil-master/libuvc-master/lib)

  find_path(LIBUSB_INCLUDE_DIR libusb-1.0/libusb.h
  PATHS E:/3.Projects/Eyes/packages/libuvc-pupil-master/libuvc-master)
else()
  find_package(PkgConfig)
  pkg_check_modules(LIBUSB libusb-1.0)
endif()

if(WIN32)
  find_library(PTHREAD_LIBRARIES pthreadVC2
        PATHS E:/3.Projects/Eyes/packages/libuvc-pupil-master/libuvc-master/pthreads-w32-2-9-1-release/Pre-built.2/lib/x64)
  find_path(PTHREAD_INCLUDE_DIR pthread.h
        PATHS E:/3.Projects/Eyes/packages/libuvc-pupil-master/libuvc-master/pthreads-w32-2-9-1-release/Pre-built.2/include)
  SET(EXTRA_DEFS "-D_TIMESPEC_DEFINED")
  add_definitions(${EXTRA_DEFS})
else()
  include(GNUInstallDirs)
  SET(CMAKE_C_FLAGS_DEBUG "-g  -DUVC_DEBUGGING")
endif()
```
In the attach files, I uploaded the zip where there is a uvc.lib that I complied. It was complied by Visual Studio 2017.  You can refer to it.
After I complied it succussfully, I coded and tried to read image according to [the cython code](https://github.com/pupil-labs/pyuvc/blob/eb2c0a04caafcb5c2f349aa76d188b098571d982/uvc.pyx#L300-L406). But I failed. It seems that some errors exist in the uvc_stream_get_frame fuction. So can you paste some codes that read images on Linux? Can it be shown by the OpenCV? Looking forward to your reply.

[libuvc-master-complied.zip](https://github.com/libuvc/libuvc/files/4872777/libuvc-master-complied.zip)

nospam2k, 2020-07-05:
@962557346 Thanks for your reply. I found libuvc-pupil-master and followed the INSTALL_WINDOWS.md completely. I didn't install LIBJPEG. I think you might want to look into that if you haven't already. I was able to compile both libusb and libuvc with no code changes using the CMAKE gui (IIRC).  I'm going to be working on coding through Java just for PTZ stuff so I won't be trying to capture images.

mcuee, 2021-05-07:
https://github.com/pupil-labs/libuvc
You will have better lucks with the fork for Windows.

mmppeegg, 2021-09-28:
A stupid question, is WinUSB driver necessary for using libuvc?

mcuee, 2021-09-28:
> A stupid question, is WinUSB driver necessary for using libuvc?

Yes.
https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows

mmppeegg, 2021-09-28:
> https://github.com/pupil-labs/libuvc
> You will have better lucks with the above libuvc fork for Windows.
Hi mcuee,
Thanks for so fast replying! I've done built successfully the pupil-labs/libuvc, and I'm trying use the libuvc.dll/libuvc.lib. But I always failed on uvc_open(). The uvc_find_device() is success by input my plugging camera's PID/VID.
Can you hint the possible losing? Thank you so much!

dmikushin, 2021-09-28:
Usually, the device does not open due to insufficient access permissions of a regular user. Please try as Administrator.

mcuee, 2021-09-29:
> Hi mcuee, Thanks for so fast replying! I've done built successfully the pupil-labs/libuvc, and I'm trying use the libuvc.dll/libuvc.lib. But I always failed on uvc_open(). The uvc_find_device() is [...]

Did you install WinUSB driver?
https://github.com/libusb/libusb/wiki/Windows#how-to-use-libusb-on-windows

mmppeegg, 2021-09-29:
> Did you install WinUSB driver? https://github.com/libusb/libusb/wiki/Windows#how-to-use-libusb-on-windows

Thank you mcuee :)
Finally I got solved with installing WinUSB. My testing is able to successfully read  my camera information via libuvc!
However I found that my camera is not visible by other applications which use cameras. Is there anyway without replacing standard usbvideo driver?

mcuee, 2021-09-29:
> Thank you mcuee :) Finally I got solved with installing WinUSB. My test program is able to successfully read my camera information via libuvc! However I found that my camera is not visible by other applications which use cameras. Is there anyway without replacing standard usbvideo driver?

That is expected if you use WinUSB driver. No way around that.

The potential workaround is to use the original Windows UVC driver and not WinUSB. But then you need to install usbdk and then change libuvc codes to use libusb usbdk backend. I guess that is too complicated and not worth doing.

Ref: https://github.com/libusb/libusb/wiki/Windows#Driver_Installation

lafith, 2022-01-19:
> @nospam2k Hello, I understand your feelings. Firstly, I will outline all the steps.
>
>     1. I downloaded the modified version ([libuvc-pupil-master](https://github.com/pupil-labs/libuvc)).
>
>     2. I [...]

Hi,
I followed the instructions and build libuvc without any error in windows. Now I am trying to use it in but the error **"fatal error C1083: Cannot open include file: 'libuvc.h': No such file or directory"** is coming.

Following is my CMakeLists.txt:
`cmake_minimum_required(VERSION 3.0.0)
project(WebCam_MicrosoftMediaFoundation VERSION 0.1.0)

include_directories(include)
file(GLOB SOURCES "src/*.cpp")

find_package(OpenCV REQUIRED)
include_directories( ${OpenCV_INCLUDE_DIRS} )

set(libuvc_DIR C:/libuvc/build/CMakeFiles)
find_package(libuvc REQUIRED)
include_directories( ${libuvc_INCLUDE_DIRS} )

include(CTest)
enable_testing()

add_executable(OpenCV_Test
                ${SOURCES})

target_link_libraries( OpenCV_Test ${OpenCV_LIBS} ${libuvc_LIBRARIES})

set(CPACK_PROJECT_NAME ${PROJECT_NAME})
set(CPACK_PROJECT_VERSION ${PROJECT_VERSION})
include(CPack)
`
CMakeList is not giving any error, but I cant include <libuvc/libvc.h>.

mcuee, 2022-05-08:
Latest build of https://github.com/pupil-labs/libuvc git head with libusb-1.0.26 (dynamic link) and pthreads-w32-2-9-1-release, using VS2019 (both 32bit and 64bit).
[pupil_labs_libuvc.zip](https://github.com/libuvc/libuvc/files/8646865/pupil_labs_libuvc.zip)

harsh-lodha, 2022-08-03:
> @nospam2k Hello, I understand your feelings. Firstly, I will outline all the steps.
>
> 1. I downloaded the modified version ([libuvc-pupil-master](https://github.com/pupil-labs/libuvc)).
> 2. I [...]

> 
Does the libuvc pupil fork support cameras which has uncompressed frame format and not mjpeg on windows?

harsh-lodha, 2022-08-03:
> > > > https://github.com/pupil-labs/libuvc
> > > > You will have better lucks with the above libuvc fork for Windows.
> > > 
> > > 
> > > Hi mcuee, Thanks for so fast replying! I've done built successfully the pupil-labs/libuvc, and I'm trying use the libuvc.dll/libuvc.lib. But I always failed on uvc_open(). The uvc_find_device() is success by input my plugging camera's PID/VID. Can you hint the possible losing? Thank you so much!
> > 
> > 
> > Did you install WinUSB driver? https://github.com/libusb/libusb/wiki/Windows#how-to-use-libusb-on-windows
> 
> Thank you mcuee :) Finally I got solved with installing WinUSB. My test program is able to successfully read my camera information via libuvc! However I found that my camera is not visible by other applications which use cameras. Is there anyway without replacing standard usbvideo driver?

Yea, I also faced the same problem. Apparently the application  only support indows default camera driver

mcuee, 2022-08-03:
> Yea, I also faced the same problem. Apparently the application only support Windows default camera driver

Unfortunately no real workaround for this one if you use WinUSB driver (or libusbk driver) to replace the Windows default camera driver. 

The potential way is to try to use usbdk backend of libusb Windows (kind of filter driver which can co-exist) with the Windows default camera driver. But then you may need to modify the libuvc code as mentioned earlier to use usbdk backend. And usbdk backend may not be as mature as WinUSB/libusbk backend.

```
libusb_context * ctx = NULL;
libusb_init(&ctx);
libusb_set_option(ctx, LIBUSB_OPTION_USE_USBDK);
```