libuvc/libuvc #146: App always hangs when uvc_stop_streaming is called, help...

Larryyuan2015, 2020-07-28:
I have checked #53,  #16 or #50 issues, its solution is already integrated on android 7.1.1 Platform.
when App switches 2 uvc cameras  about 20 times，it always hangs finally.

when it runs pthread_cond_wait(&strmh->cb_cond, &strmh->cb_mutex);
it seems to wait the signal.
I don't know how to fix it.  many thanks.

Log：

12-12 08:12:42.556  3070  3093 I libUVCCamera: [3093*UVCPreview.cpp:366:stopPreview]:UVCPreview::stopPreview 2 b= 
12-12 08:12:42.556  3070  4441 I libUVCCamera: [4441*UVCPreview.cpp:567:do_preview]:preview_thread_func:wait for all callbacks complete
12-12 08:12:42.556  3070  4441 I libuvc/stream: [4441*stream.c:1823:uvc_stop_streaming]:uvc_stop_streaming start
12-12 08:12:42.556  3070  4441 I libuvc/stream: [4441*stream.c:1934:uvc_stream_close]:uvc_stream_close 0 strmh->running=1
12-12 08:12:42.556  3070  4441 I libuvc/stream: [4441*stream.c:1855:uvc_stream_stop]: uvc_stream_stop strmh->running=0
12-12 08:12:42.556  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 0 (0xc77fc624)
12-12 08:12:42.556  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 1 (0xc77fc824)
12-12 08:12:42.558  3070  3070 V MainActivity: onSurfaceTextureUpdated mCaptureState:0 tid=1
12-12 08:12:42.558  3070  3070 V MainActivity: onSurfaceTextureUpdated mEncoder:null tid=1
12-12 08:12:42.565  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 2 (0xc77fca24)
12-12 08:12:42.565  3070  4440 I libuvc/stream: [4440*stream.c:977:_uvc_stream_callback]: _uvc_stream_callback...transfer->status=0
12-12 08:12:42.565  3070  4440 I libuvc/stream: [4440*stream.c:994:_uvc_stream_callback]:isochronous...transfer->num_iso_packets=32
12-12 08:12:42.565  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 3 (0xc77fcc24)
12-12 08:12:42.565  3070  4440 I libuvc/stream: [4440*stream.c:1022:_uvc_stream_callback]: 1 _uvc_delete_transfer...transfer->status=0
12-12 08:12:42.565  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 4 (0x0)
12-12 08:12:42.565  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 5 (0x0)
12-12 08:12:42.565  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 6 (0xccc88e24)
12-12 08:12:42.567  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 7 (0xde8e7024)
12-12 08:12:42.570  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 8 (0xde8e7224)
12-12 08:12:42.571  3070  4441 I libuvc/stream: [4441*stream.c:1861:uvc_stream_stop]: uvc_stream_stop libusb_cancel_transfer 9 (0xde8e7424)
12-12 08:12:42.572  3070  4441 I libuvc/stream: [4441*stream.c:1884:uvc_stream_stop]: uvc_stream_stop LIBUVC_NUM_TRANSFER_BUFS =10
12-12 08:12:42.572  3070  4441 I libuvc/stream: [4441*stream.c:1893:uvc_stream_stop]: pthread_cond_wait start i=0 
12-12 08:12:42.573  3070  4440 I libuvc/stream: [4440*stream.c:644:_uvc_delete_transfer]: Freeing transfer 2 (0xc77fca24)
12-12 08:12:42.573  3070  4440 I libuvc/stream: [4440*stream.c:659:_uvc_delete_transfer]: _uvc_delete_transfer finished
12-12 08:12:42.573  3070  4440 I libuvc/stream: [4440*stream.c:1028:_uvc_stream_callback]: 2 _uvc_delete_transfer...transfer->status=0
12-12 08:12:42.573  3070  4441 I libuvc/stream: [4441*stream.c:1899:uvc_stream_stop]:  pthread_cond_wait end i=0 

12-12 08:12:42.573  3070  4441 I libuvc/stream: [4441*stream.c:1893:uvc_stream_stop]: pthread_cond_wait start i=0 
12-12 08:12:42.573  3070  4442 I libuvc/stream: [4442*stream.c:1681:_uvc_user_caller]:_uvc_user_caller 2 strmh->running = 0 
12-12 08:12:42.573  3070  4440 I libuvc/stream: [4440*stream.c:977:_uvc_stream_callback]: _uvc_stream_callback...transfer->status=3
12-12 08:12:42.573  3070  4440 I libuvc/stream: [4440*stream.c:1022:_uvc_stream_callback]: 1 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.573  3070  4440 I libuvc/stream: [4440*stream.c:644:_uvc_delete_transfer]: Freeing transfer 3 (0xc77fcc24)
12-12 08:12:42.573  3070  4440 I libuvc/stream: [4440*stream.c:659:_uvc_delete_transfer]: _uvc_delete_transfer finished
12-12 08:12:42.573  3070  4440 I libuvc/stream: [4440*stream.c:1028:_uvc_stream_callback]: 2 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:977:_uvc_stream_callback]: _uvc_stream_callback...transfer->status=3
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:1022:_uvc_stream_callback]: 1 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:644:_uvc_delete_transfer]: Freeing transfer 1 (0xc77fc824)
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:659:_uvc_delete_transfer]: _uvc_delete_transfer finished
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:1028:_uvc_stream_callback]: 2 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:977:_uvc_stream_callback]: _uvc_stream_callback...transfer->status=3
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:1022:_uvc_stream_callback]: 1 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:644:_uvc_delete_transfer]: Freeing transfer 6 (0xccc88e24)
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:659:_uvc_delete_transfer]: _uvc_delete_transfer finished
12-12 08:12:42.574  3070  4440 I libuvc/stream: [4440*stream.c:1028:_uvc_stream_callback]: 2 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.575  3070  4440 I libuvc/stream: [4440*stream.c:977:_uvc_stream_callback]: _uvc_stream_callback...transfer->status=3
12-12 08:12:42.575  3070  4440 I libuvc/stream: [4440*stream.c:1022:_uvc_stream_callback]: 1 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.575  3070  4440 I libuvc/stream: [4440*stream.c:644:_uvc_delete_transfer]: Freeing transfer 7 (0xde8e7024)
12-12 08:12:42.575  3070  4440 I libuvc/stream: [4440*stream.c:659:_uvc_delete_transfer]: _uvc_delete_transfer finished
12-12 08:12:42.575  3070  4440 I libuvc/stream: [4440*stream.c:1028:_uvc_stream_callback]: 2 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.575  3070  4441 I libuvc/stream: [4441*stream.c:1899:uvc_stream_stop]:  pthread_cond_wait end i=0 

12-12 08:12:42.575  3070  4441 I libuvc/stream: [4441*stream.c:1893:uvc_stream_stop]: pthread_cond_wait start i=0 
12-12 08:12:42.576  3070  4440 I libuvc/stream: [4440*stream.c:977:_uvc_stream_callback]: _uvc_stream_callback...transfer->status=3
12-12 08:12:42.576  3070  4440 I libuvc/stream: [4440*stream.c:1022:_uvc_stream_callback]: 1 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.576  3070  4440 I libuvc/stream: [4440*stream.c:644:_uvc_delete_transfer]: Freeing transfer 8 (0xde8e7224)
12-12 08:12:42.576  3070  4440 I libuvc/stream: [4440*stream.c:659:_uvc_delete_transfer]: _uvc_delete_transfer finished
12-12 08:12:42.576  3070  4440 I libuvc/stream: [4440*stream.c:1028:_uvc_stream_callback]: 2 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.576  3070  4441 I libuvc/stream: [4441*stream.c:1899:uvc_stream_stop]:  pthread_cond_wait end i=0 

12-12 08:12:42.576  3070  4441 I libuvc/stream: [4441*stream.c:1893:uvc_stream_stop]: pthread_cond_wait start i=0 
12-12 08:12:42.577  3070  4440 I libuvc/stream: [4440*stream.c:977:_uvc_stream_callback]: _uvc_stream_callback...transfer->status=3
12-12 08:12:42.577  3070  4440 I libuvc/stream: [4440*stream.c:1022:_uvc_stream_callback]: 1 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.577  3070  4440 I libuvc/stream: [4440*stream.c:644:_uvc_delete_transfer]: Freeing transfer 9 (0xde8e7424)
12-12 08:12:42.577  3070  4440 I libuvc/stream: [4440*stream.c:659:_uvc_delete_transfer]: _uvc_delete_transfer finished
12-12 08:12:42.577  3070  4440 I libuvc/stream: [4440*stream.c:1028:_uvc_stream_callback]: 2 _uvc_delete_transfer...transfer->status=3
12-12 08:12:42.577  3070  4441 I libuvc/stream: [4441*stream.c:1899:uvc_stream_stop]:  pthread_cond_wait end i=0 

12-12 08:12:42.578  3070  4441 I libuvc/stream: [4441*stream.c:1893:uvc_stream_stop]: pthread_cond_wait start i=0

Larryyuan2015, 2020-07-28:
struct libusb_transfer *transfers[LIBUVC_NUM_TRANSFER_BUFS];

LIBUVC_NUM_TRANSFER_BUFS is 10， transfer[4] and transfer[5] are freed, but one transfer[0] is still not deleted.

12-12 08:12:39.564  3070  4440 I libuvc/stream: [4440*stream.c:977:_uvc_stream_callback]: _uvc_stream_callback...transfer->status=1
12-12 08:12:39.564  3070  4440 I libuvc/stream: [4440*stream.c:1022:_uvc_stream_callback]: 1 _uvc_delete_transfer...transfer->status=1
12-12 08:12:39.564  3070  4440 I libuvc/stream: [4440*stream.c:644:_uvc_delete_transfer]: Freeing transfer 5 (0xccc88c24)
12-12 08:12:39.564  3070  4440 I libuvc/stream: [4440*stream.c:659:_uvc_delete_transfer]: _uvc_delete_transfer finished
12-12 08:12:39.565  3070  4440 I libuvc/stream: [4440*stream.c:1028:_uvc_stream_callback]: 2 _uvc_delete_transfer...transfer->status=1
12-12 08:12:39.565  3070  4442 I libuvc/stream: [4442*stream.c:1676:_uvc_user_caller]:_uvc_user_caller strmh->running = 1

12-12 08:12:39.882  3070  4440 I libuvc/stream: [4440*stream.c:1022:_uvc_stream_callback]: 1 _uvc_delete_transfer...transfer->status=1
12-12 08:12:39.882  3070  4440 I libuvc/stream: [4440*stream.c:644:_uvc_delete_transfer]: Freeing transfer 4 (0xc77fce24)
12-12 08:12:39.883  3070  4440 I libuvc/stream: [4440*stream.c:659:_uvc_delete_transfer]: _uvc_delete_transfer finished
12-12 08:12:39.883  3070  4440 I libuvc/stream: [4440*stream.c:1028:_uvc_stream_callback]: 2 _uvc_delete_transfer...transfer->status=1
12-12 08:12:39.883  3070  4442 I libuvc/stream: [4442*stream.c:1676:_uvc_user_caller]:_uvc_user_caller strmh->running = 1

Larryyuan2015, 2020-07-28:
when pthread_cond_wait runs and wait the last transfer[0] to be deleted, 
I don't know why the  _uvc_stream_callback function is not called.

Larryyuan2015, 2020-07-31:
when the app happens to hang，soon it will hang again at the first switch between 2 uvc cameras,  even if the app reinstalled again.
only when restart the android system , the app needs switch 2 uvc cameras about 20 time to hang.
how to analyze the issue?

PeterPeterPan, 2020-10-05:
Which version of libuvc are you using?
I think https://github.com/libuvc/libuvc/issues/91 may helpful for you.