libuvc/libuvc #148: Support for iOS

iosfitness, 2020-08-10:
Can anyone please redirect me to links for implementing this on iOS. I plan to connect Logitech webcam c930e

chrisspiegl, 2020-08-13:
I was looking for the same. Ability to compile this for iOS and use it to create a UVC compatible live streaming application.

iosfitness, 2020-08-13:
@chrisspiegl Did u manage to find any workaround for this?

chrisspiegl, 2020-08-15:
@iosfitness I did not, I am hoping to see whether or not this is compatible at first.

iosfitness, 2020-10-10:
@chrisspiegl Did u manage to look into this? Also, it would be great if you could please suggest any alternatives for now.

chrisspiegl, 2020-10-10:
@iosfitness I had to stop looking into it since it was too big of a project/idea. I have not found any solution that I liked thus far.

iosfitness, 2020-10-12:
@chrisspiegl Could you please share the solutions/links you explored so far ?. 
I will go through them and check If I can build anything  using them

chrisspiegl, 2020-10-12:
@iosfitness The only solution I found in my research is this: https://www.youtube.com/watch?v=lenGIGSnB0M 
But that is a extra device that I was not prepared to purchase. My idea was that it would be great to be able to use a CamLink or similar HDMI capture card with the iPad for Live Streaming, however, even the device shown in the video does not provide that with good app compatibility.
So I had to drop it and resort back to using a normal MacBook / Windows Laptop for Live Streaming from there.
The development process — though enticing — was just too much of a hasstle.
Last resort, I am waiting for Apple to make iOS or at least iPad OS compatible with UVC capture devices… that would be an incredible step into the direction of making the iPad a really amazing device.
Cheers, hope that helps.

iosfitness, 2020-10-14:
@chrisspiegl 
Thanks for sharing the knowledge

Few inputs from my end :
I did some exploring initially and landed up trying out https://developer.apple.com/documentation/externalaccessory. However, I think we will need to have information on the UISupportedExternalAccessoryProtocols to make this functional.
For this, I also purchased https://www.apple.com/in/shop/product/MK0W2ZM/A/lightning-to-usb-3-camera-adapter and https://www.amazon.in/Logitech-Widescreen-Calling-Recording-960-000764/dp/B006JH8T3S

This solution looked promising but unfortunately didn't work for me.

Still looking for alternatives. Will keep you posted if any find anything

chrisspiegl, 2020-10-14:
Thanks for sharing your information. I tried the external webcam with an iPad Pro which has USB-C so I didn't need the adapter you also purchased. However with no luck on all fronts. Hopefully Apple would just start supporting UVC capture devices at some point. All their MacBooks support them… why not the iOS versions? I don't know… but I still have hope.

mcuee, 2021-08-06:
It is not possible to build libuvc under iOS as libusb is no supported under iOS. So you have to look elsewhere.

RainyLave, 2024-07-07:
With iokit on Xcode we can build a static lib! now I've built libuvc libusb and libjpeg for archive arm64 iOS and amrv7 iOS, that's works!

mcuee, 2024-07-07:
Not very useful. 
libusb does not work under iOS/iPadOS unless you jailbreak the device. It is not recommended to jailbreak an iOS/iPadOS devices.

akmadov, 2024-11-19:
> With iokit on Xcode we can build a static lib! now I've built libuvc libusb and libjpeg for archive arm64 iOS and amrv7 iOS, that's works! you can find them on my reporsity!

Hi there! Do you think that the LiveView app can be made to work on iOS 12.5.1? Thanks!

mcuee, 2026-09-06:
Close this as not planned as iOS is not officially supported by libusb.

closed 2026-09-06 by mcuee