libuvc/libuvc #149: Problem with static linking

jwlodek, 2020-08-17:
I would like to statically link libuvc into an executable, however, whenever I try I get undefined reference errors w.r.t. libusb. I am using a basic example similar to the one given here, that just lists attached cameras + serial numbers. Here is the contents of my folder:

```
jwlodek@ubuntu:/epics/src/support/areaDetector/ADUVC/uvcSupport/cameraDetector$ ls
libusb.a  libuvc  libuvc.a  Makefile  README.md  utlist.h  uvc_locater.cpp
```

Here is the makefile:

```
all:
	g++ -L. uvc_locater.cpp -o uvc_locater -lusb -luvc
clean:
	rm uvc_locater
```

Then, when I run `make`, I get many similar errors along the lines of:

```
/usr/bin/ld: ./libuvc.a(init.c.o): in function `_uvc_handle_events':
init.c:(.text+0x27): undefined reference to `libusb_handle_events_completed'
```

Obviously something is not linking correctly to libusb, for the static build. For the non-static build, I seemingly have no such issues.

openastroproject, 2020-08-17:
Try reversing the order of the libraries, so they're "-luvc -lusb"?

jwlodek, 2020-08-17:
I tried that and got the same set of errors. I also tried removing `libusb.a` to try and have the compiler use the system version, and that didn't work as well. Removing `-lusb` altogether also just shows the same errors.

jwlodek, 2020-08-17:
I got it. Seems that there are two different `libusb`'s  out there, and using `grep` on the CMake logs, I found that `libuvc` is linking against `libusb-1.0.so`, not `libusb.so`. Once I linked against this the build completed OK. Perhaps this should be noted somewhere in the documentation, or in the CMake output to list which version is linked?