libuvc/libuvc #150: CMake Error, target "LibUSB::LibUSB" was not found.

PeterPeterPan, 2020-08-18:
Hi all,
In my windows 10 laptop, I install "libusb 1.0.23" and "cmake  3.17.2" to Anaconda
then follow the step:
git clone https://github.com/libuvc/libuvc
cd libuvc
mkdir build
cd build
cmake ..

when I input: cmake ..
I got the error message

> (base) C:\Users\t1994\libuvc\build>cmake ..
> -- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
> CMake Warning at CMakeLists.txt:51 (message):
>   JPEG not found.  libuvc will not support JPEG decoding.
> 
> -- Configuring done
> CMake Error at CMakeLists.txt:66 (add_library):
>   Target "uvc" links to target "LibUSB::LibUSB" but the target was not found.
>   Perhaps a find_package() call is missing for an IMPORTED target, or an
>   ALIAS target is missing?
> 
> CMake Error at CMakeLists.txt:66 (add_library):
>   Target "uvc" links to target "LibUSB::LibUSB" but the target was not found.
>   Perhaps a find_package() call is missing for an IMPORTED target, or an
>   ALIAS target is missing?
> 
> CMake Error at CMakeLists.txt:66 (add_library):
>   Target "uvc" links to target "LibUSB::LibUSB" but the target was not found.
>   Perhaps a find_package() call is missing for an IMPORTED target, or an
>   ALIAS target is missing?
> 
> CMake Error at CMakeLists.txt:66 (add_library):
>   Target "uvc" links to target "LibUSB::LibUSB" but the target was not found.
>   Perhaps a find_package() call is missing for an IMPORTED target, or an
>   ALIAS target is missing?
> 
> CMake Error at CMakeLists.txt:83 (add_library):
>   Target "uvc_static" links to target "LibUSB::LibUSB" but the target was not
>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
>   an ALIAS target is missing?
> 
> CMake Error at CMakeLists.txt:83 (add_library):
>   Target "uvc_static" links to target "LibUSB::LibUSB" but the target was not
>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
>   an ALIAS target is missing?
> 
> CMake Error at CMakeLists.txt:83 (add_library):
>   Target "uvc_static" links to target "LibUSB::LibUSB" but the target was not
>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
>   an ALIAS target is missing?
> 
> CMake Error at CMakeLists.txt:83 (add_library):
>   Target "uvc_static" links to target "LibUSB::LibUSB" but the target was not
>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
>   an ALIAS target is missing?
> 
> CMake Error at CMakeLists.txt:66 (add_library):
>   Target "uvc" links to target "LibUSB::LibUSB" but the target was not found.
>   Perhaps a find_package() call is missing for an IMPORTED target, or an
>   ALIAS target is missing?
> 
> CMake Error at CMakeLists.txt:83 (add_library):
>   Target "uvc_static" links to target "LibUSB::LibUSB" but the target was not
>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
>   an ALIAS target is missing?
> 
> -- Generating done
> CMake Generate step failed.  Build files cannot be regenerated correctly.

How can I solve this problem?
thank for your help.

closed 2020-08-25 by PeterPeterPan

WallyChiu, 2020-08-27:
Hi PeterPeterPan,

I have the same issue. Would you tell me how did you solve the problem? Thnaks.

PeterPeterPan, 2020-08-31:
Hi Wally,
I didn't solve this problem, so I trying to use visual studio 2019 with libuvc, too.
follow this way( you can reference from issue#100 ):
https://github.com/pupil-labs/libuvc
https://github.com/pupil-labs/libuvc/blob/master/INSTALL_WINDOWS.md 

I can compile the sample code now, but still some problem happen.
The sample can detect my device by "uvc_find_device", but "uvc_open" return error (-5).

PeterPeterPan, 2020-08-31:
update:
I used Zadig to change the driver of my device from WinUSB to libusbK.
Now error become Access denied (-3)

PeterPeterPan, 2020-09-01:
update:
Using Zadig to change the driver of my device from Vendor-driver to libusb-win32(v1.2.6.0).

Now, uvc_start_streaming still not work, but I can use "uvc_set_ctrl" to control Units.

PeterPeterPan, 2020-09-04:
update:
If your device only with video, it's working now, but not perfect.

fork:

> https://github.com/pupil-labs/libuvc
> https://github.com/pupil-labs/libuvc/blob/master/INSTALL_WINDOWS.md

note:

> Visual studio 2019 (Win10)
> Cmake (3.18.2 x64)
> Opencv3.4.11 (vc15)
> Change device Composite Parent to libusbK. (using Zadig)

Device_1: BisonCam,NB Pro, only video.
Device_2: AVerVision M15W, with video&audio.

Using the test.c to work in UVC_FRAME_FORMAT_YUYV, only Device_1 show image, Device_2 freeze & do noting.
But broken as the step "uvc_stop_streaming" both two, keep trying...