libuvc/libuvc #153: uvc_start_streaming not work as expected

stang2019, 2020-08-25:
I hooked two webcams, either will work if I start with first chosen, if I do like this, select first camera, start streaming, the callback will receive the video frames and it works normally, then I stop streaming using uvc_stop_streaming, open the second webcam to get its handle (successful), call uvc_get_stream_ctrl_format_size to set the resolution and FPS (successful), then start streaming with res = uvc_start_streaming(devh, &ctrl, cb, this, 0), the res was UVC_SUCCESS, but there is no video frame delivered to cb callback.

This is the stream func and the callback:
assume the devh is the handle of new opening device:

void cb(uvc_frame_t *frame, void* env)
{
    Camera* pCam = reinterpret_cast<Camera*>(env);
    ....
}

int Camera::startStream()
{
    if (isStreaming) return true;
    //1. create the control

    auto res = uvc_get_stream_ctrl_format_size(devh, &ctrl, curFrameFormat, videoWidth, videoHeight, fps);
    if (res != UVC_SUCCESS)
    {
        loggerEx(LOG_DEBUG, "Start Streaming, Failed to matching format:%s", uvc_strerror(res));
        return res;
    }
    //Matched, now start streaming
    res = uvc_start_streaming(devh, &ctrl, cb, this, 0);
    if (res != UVC_SUCCESS)
    {
        loggerEx(LOG_DEBUG, "Start Streaming, Failed to start streaming:%s", uvc_strerror(res));
        return res;
    }
    isStreaming = true;
    return res;
}

JoelStienlet, 2020-08-29:
I had a similar problem, I solved it by adjusting the buffer size in the kernel:
`echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb`
Hope this helps.

stang2019, 2020-08-29:
Thanks, I’ll try it out.

-Steven

rsnGON, 2020-11-28:
I had a similar problem,  i use Product: PureThermal (fw:v1.3.0) on Ubuntu 16

` Linux version 4.15.0-123-generic (buildd@lcy01-amd64-027) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)) #126~16.04.1-Ubuntu SMP Wed Oct 21 13:48:05 UTC 2020`

dmesg out is :
`[271680.726994] usb 1-1.5: USB disconnect, device number 38
[271682.918791] usb 1-1.5: new full-speed USB device number 39 using ehci-pci
[271683.030094] usb 1-1.5: New USB device found, idVendor=1e4e, idProduct=0100
[271683.030099] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[271683.030103] usb 1-1.5: Product: PureThermal (fw:v1.3.0)
[271683.030106] usb 1-1.5: Manufacturer: GroupGets
[271683.030109] usb 1-1.5: SerialNumber: 0009800b-5113-3437-3335-373400000000
[271683.031023] uvcvideo: Found UVC 1.00 device PureThermal (fw:v1.3.0) (1e4e:0100)
[271683.031905] uvcvideo 1-1.5:1.0: Entity type for entity Extension 3 was not initialized!
[271683.031910] uvcvideo 1-1.5:1.0: Entity type for entity Processing 2 was not initialized!
[271683.031913] uvcvideo 1-1.5:1.0: Entity type for entity Extension 4 was not initialized!
[271683.031916] uvcvideo 1-1.5:1.0: Entity type for entity Extension 5 was not initialized!
[271683.031918] uvcvideo 1-1.5:1.0: Entity type for entity Extension 6 was not initialized!
[271683.031921] uvcvideo 1-1.5:1.0: Entity type for entity Extension 7 was not initialized!
[271683.031924] uvcvideo 1-1.5:1.0: Entity type for entity Extension 21 was not initialized!
[271683.031927] uvcvideo 1-1.5:1.0: Entity type for entity Extension 254 was not initialized!
[271683.031930] uvcvideo 1-1.5:1.0: Entity type for entity Camera 1 was not initialized!
[271683.032335] input: PureThermal (fw:v1.3.0): PureTh as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0/input/input51
[271684.646076] usb 1-1.5: USB disconnect, device number 39
[271684.906787] usb 1-1.5: new full-speed USB device number 40 using ehci-pci
[271685.018095] usb 1-1.5: New USB device found, idVendor=1e4e, idProduct=0100
[271685.018101] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[271685.018104] usb 1-1.5: Product: PureThermal (fw:v1.3.0)
[271685.018107] usb 1-1.5: Manufacturer: GroupGets
[271685.018109] usb 1-1.5: SerialNumber: 0009800b-5113-3437-3335-373400000000
[271685.018749] uvcvideo: Found UVC 1.00 device PureThermal (fw:v1.3.0) (1e4e:0100)
[271685.019785] uvcvideo 1-1.5:1.0: Entity type for entity Extension 3 was not initialized!
[271685.019789] uvcvideo 1-1.5:1.0: Entity type for entity Processing 2 was not initialized!
[271685.019792] uvcvideo 1-1.5:1.0: Entity type for entity Extension 4 was not initialized!
[271685.019795] uvcvideo 1-1.5:1.0: Entity type for entity Extension 5 was not initialized!
[271685.019798] uvcvideo 1-1.5:1.0: Entity type for entity Extension 6 was not initialized!
[271685.019801] uvcvideo 1-1.5:1.0: Entity type for entity Extension 7 was not initialized!
[271685.019803] uvcvideo 1-1.5:1.0: Entity type for entity Extension 21 was not initialized!
[271685.019806] uvcvideo 1-1.5:1.0: Entity type for entity Extension 254 was not initialized!
[271685.019809] uvcvideo 1-1.5:1.0: Entity type for entity Camera 1 was not initialized!
[271685.020011] input: PureThermal (fw:v1.3.0): PureTh as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0/input/input52
[271685.971067] usb 1-1.5: usbfs: usb_submit_urb returned -28`

`echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb`  this was not worked

pandeypradyumn, 2022-06-30:
any idea how can i solve it on windows

AaronSomeone, 2024-03-21:
> any idea how can i solve it on windows

Did you figure it out?

MachineL102, 2024-08-25:
i have same problem, can anyone help（I was forced to work overtime by my boss）?  very thanks~~~~~~~~~~~

MachineL102, 2024-08-25:
I solved this by inserted two usb camera separetely on two usb bus