libuvc/libuvc #156: "JpegPKG" cmake files library not found on MacOS

RedHoodedWraith, 2020-09-18:
Hi,

I am attempting to import the _LibUVC_ Library in a CMakeLists.txt for a C++ project. I am running MacOS 10.15.6

The full error message is outlined below. In a nutshell, the error I am getting is that CMake is having trouble finding the following files from the `JpegPKG` library:
- JpegPkgConfig.cmake
- jpegpkg-config.cmake

I have installed `ibjpeg` by using `brew install libjpeg` however that did not seem to work. I have made many attempts to search for "JpegPKG" or the missing files in question however I keep getting redirected to the same answers regarding installing ROS, which I cannot do on my system. 

Does anyone know where I can find the source code library for `JpegPKG` that has the two cmake files listed above? I don't think I need this Jpeg package for my build however it seems LibUVC requires it. 

**This is the error message I am getting:**
```
/Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /Users/rowan/Development/JourneyEye/Recording_Prototypes/Camera_Configuration_Prototype_C++
CMake Error at /Users/rowan/libuvc/LibUVCConfig.cmake:13 (find_package):
  By not providing "FindJpegPkg.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "JpegPkg", but
  CMake did not find one.

  Could not find a package configuration file provided by "JpegPkg" with any
  of the following names:

    JpegPkgConfig.cmake
    jpegpkg-config.cmake

  Add the installation prefix of "JpegPkg" to CMAKE_PREFIX_PATH or set
  "JpegPkg_DIR" to a directory containing one of the above files.  If
  "JpegPkg" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:6 (find_package)
```

**This is the CMakeList.txt I have configured for my C++ program:**
```cmake_minimum_required(VERSION 3.17)
project(Camera_Configuration_Prototype_C__)

set(CMAKE_CXX_STANDARD 14)

find_package(LibUVC REQUIRED)
set(SOURCE_FILES main.cpp)
add_executable(${PROJECT_NAME} ${SOURCE_FILES})
include_directories(${LibUVC_INCLUDE_DIRS})
target_link_libraries(${PROJECT_NAME} ${LibUVC_LIBS})```

And the issue seems to stem from the file [libuvcConfig.cmake](/libuvc/blob/master/libuvcConfig.cmake) which was called by my CMakeLists.txt file from above.

RedHoodedWraith, 2020-09-18:
When rebuilding the LibUVC Library using `cmake build ..` when inside the build directory within the directory with the library's source code, I did notice this error message:
```
-- libusb-1.0 found using pkgconfig
-- Found JPEG: /usr/local/lib/libjpeg.dylib (found version "80") 
-- Found JPEG library using standard module
CMake Warning (dev) at /usr/local/Cellar/cmake/3.18.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (JPEG) does
  not match the name of the calling package (JpegPkg).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindJpegPkg.cmake:58 (find_package_handle_standard_args)
  CMakeLists.txt:45 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found JPEG: /usr/local/lib/libjpeg.dylib  
-- Building libuvc with JPEG support.
-- Components requested: opencv_core;opencv_highgui
-- Configuring done
CMake Error in CMakeLists.txt:
  Cannot use features from non-enabled language CXX

CMake Generate step failed.  Build files cannot be regenerated correctly.
```

mcuee, 2021-08-06:
Working fine here with Homebrew and macOS Big Sur (11.5) on my Mac Mini M1.

```

libuvc/buildm1 on  master ❯ cmake ..
-- The C compiler identification is AppleClang 12.0.5.12050022
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- No build type selected, default to Release
-- No target type selected, default to both shared and static library
-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.24
-- libusb-1.0 found using pkgconfig
-- Found JPEG: /opt/homebrew/lib/libjpeg.dylib (found version "80") 
-- Found JPEG library using standard module
CMake Warning (dev) at /opt/homebrew/Cellar/cmake/3.21.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (JPEG) does
  not match the name of the calling package (JpegPkg).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindJpegPkg.cmake:58 (find_package_handle_standard_args)
  CMakeLists.txt:45 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found JPEG: /opt/homebrew/lib/libjpeg.dylib  
-- Building libuvc with JPEG support.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mcuee/build/libusb/applications/libuvc/buildm1
libuvc/buildm1 on  master [?] via △ v3.21.1 ❯ make
[  4%] Building C object CMakeFiles/uvc.dir/src/ctrl.c.o
[  9%] Building C object CMakeFiles/uvc.dir/src/ctrl-gen.c.o
[ 13%] Building C object CMakeFiles/uvc.dir/src/device.c.o
[ 18%] Building C object CMakeFiles/uvc.dir/src/diag.c.o
[ 22%] Building C object CMakeFiles/uvc.dir/src/frame.c.o
[ 27%] Building C object CMakeFiles/uvc.dir/src/init.c.o
[ 31%] Building C object CMakeFiles/uvc.dir/src/stream.c.o
[ 36%] Building C object CMakeFiles/uvc.dir/src/misc.c.o
[ 40%] Building C object CMakeFiles/uvc.dir/src/frame-mjpeg.c.o
[ 45%] Linking C shared library libuvc.dylib
[ 45%] Built target uvc
[ 50%] Building C object CMakeFiles/example.dir/src/example.c.o
/Users/mcuee/build/libusb/applications/libuvc/src/example.c:26:74: warning: cast to smaller integer type 'int' from 'void *' [-Wvoid-pointer-to-int-cast]
    frame->frame_format, frame->width, frame->height, frame->data_bytes, (int) ptr);
                                                                         ^~~~~~~~~
1 warning generated.
[ 54%] Linking C executable example
[ 54%] Built target example
[ 59%] Building C object CMakeFiles/uvc_static.dir/src/ctrl.c.o
[ 63%] Building C object CMakeFiles/uvc_static.dir/src/ctrl-gen.c.o
[ 68%] Building C object CMakeFiles/uvc_static.dir/src/device.c.o
[ 72%] Building C object CMakeFiles/uvc_static.dir/src/diag.c.o
[ 77%] Building C object CMakeFiles/uvc_static.dir/src/frame.c.o
[ 81%] Building C object CMakeFiles/uvc_static.dir/src/init.c.o
[ 86%] Building C object CMakeFiles/uvc_static.dir/src/stream.c.o
[ 90%] Building C object CMakeFiles/uvc_static.dir/src/misc.c.o
[ 95%] Building C object CMakeFiles/uvc_static.dir/src/frame-mjpeg.c.o
[100%] Linking C static library libuvc.a
[100%] Built target uvc_static
```

mcuee, 2021-08-06:
But if your cmake version is old, maybe you can try the tip here: https://github.com/libuvc/libuvc/issues/180

closed 2026-09-06 by mcuee