libuvc/libuvc #158: PTZ control issues

jwlodek, 2020-09-29:
I've been trying to get PTZ control working with a camera using libuvc and I'm running into some issues. First, the MAX, and MIN get operations for pan/tilt don't seem to give accurate numbers, and no matter what input I give as pan or tilt to the `abs` version of the function I get a UVC_PIPE error. When I use the relative version, I can only pass 1 or -1 as a value, which just sets the camera direction, however, I can't seem to figure out how to stop the camera. If there was a way, I could use the speed + time to stop the camera at any position. Has anyone figured out how to do this? Thanks.

jwlodek, 2020-09-29:
I figured out how to work with the relative pan/tilt. Simply call the set function again and pass 0 as the value. It would be nice if there was a function that could internally on a relative scale map from max pan to min pan etc. I still can't seem to get the absolute pan/tilt to do anything.

r00t-, 2022-03-09:
are you sure your camera actually supports absolute positioning? is it possible with the manufacturer's software? it's conceivable that there simply is no position feedback, because that makes the hardware cheaper to build (less sensors), and it can still be sold as pan/tilt.