libuvc/libuvc #159: Compilation fails on older linux distributions

jwlodek, 2020-10-16:
Compilation of latest libuvc fails on older linux distributions (Debian 7) with the following error:
```
[ 38%] Building C object CMakeFiles/uvc.dir/src/stream.c.o                                                            │
/home/jwlodek/usbptz-build/support/areaDetector/ADUVC/uvcSupport/libuvc/src/stream.c: In function ‘uvc_stream_start’: │
/home/jwlodek/usbptz-build/support/areaDetector/ADUVC/uvcSupport/libuvc/src/stream.c:1142:46: error: dereferencing poi│
nter to incomplete type                                                                                               │
make[2]: *** [CMakeFiles/uvc.dir/src/stream.c.o] Error 1                                                              │
make[1]: *** [CMakeFiles/uvc.dir/all] Error 2                                                                         │
make: *** [all] Error 2
```

It seems this issue was introduced with commit 86ad9640392aaee873ab76e61d2f4f121136ca1a. Rolling back to once commit prior it compiles OK.

Since older distributions are no longer actively supported this is likely not a problem - we still run older versions on non-networked devices, but we will simply roll back to the previous version for now.

openastroproject, 2020-10-16:
Perhaps this section of code at around line 1142 in stream.c:

```
        struct libusb_ss_endpoint_companion_descriptor *ep_comp = 0;
        libusb_get_ss_endpoint_companion_descriptor(NULL, endpoint, &ep_comp);
        if (ep_comp)
        {
          endpoint_bytes_per_packet = ep_comp->wBytesPerInterval;
          libusb_free_ss_endpoint_companion_descriptor(ep_comp);
          break;
        }
        else
```

could be wrapped in
```
#if defined(LIBUSB_API_VERSION) && (LIBUSB_API_VERSION >=  0x01000102)
...
#endif
```

I think that would leave the old functionality in place whilst removing the superspeed code that doesn't exist in earlier releases of libusb?

jwlodek, 2020-10-17:
I think that would be a good solution. I'll try it at work next week and if it works I'll make a pull request.

mcuee, 2026-09-06:
Close this old issue now.

closed 2026-09-06 by mcuee