libuvc/libuvc #167: uvc_exit() hangs for ever on internal thread join.

mrspirytus, 2021-01-21:
Hello,

I am not sure if the problem is libuvc or libusb. I will start here and if this needs to be moved or linked to libusb I will do so. Please let me know If I should create libusb ticket.

I have a daemon process that monitors cameras and publishes captured frames. Occasionally I can send `kill -INT $PID` to reload configurations. When SIGINT is delivered and the application quits and restarts itself in main() in the same fashion as the initial run. Everything is running fine and I can restart it multiple times without the server ever actually quitting. Where it gets interesting is that then I fork() myself things get bad on shutdown. So fork is the very first thing performed. So all initialization in the child process. So after I start my daemon process on the very first time and try to send `kill -INT $PID` I see sig handler fires and starts the shutdown / reload process. In this case, the shutdown never finishes. This is reproducible 100% of the time. each time I have to kill -KILL, restart fresh and send kill -INT.
After attaching gdb to the "hanging" process here is what is on the stack:
```
(gdb) where
#0  __pthread_clockjoin_ex
    (threadid=140414809589504, thread_return=0x0, clockid=<optimized out>, abstime=<optimized out>, block=<optimized out>)
    at pthread_join_common.c:145
#1  0x00007fb4ee4a95b1 in  () at /lib/x86_64-linux-gnu/libusb-1.0.so.0
#2  0x00007fb4ee4a7905 in  () at /lib/x86_64-linux-gnu/libusb-1.0.so.0
#3  0x00007fb4ee49f207 in libusb_exit () at /lib/x86_64-linux-gnu/libusb-1.0.so.0
#4  0x00007fb4efa20128 in uvc_exit () at /lib/x86_64-linux-gnu/libuvc.so.0
#5  0x00005555770f7305 in UVCDevice::cleanup() (this=0x555578877310)
```

I also enabled `LIBUSB_DEBUG=4` so I am attaching libusb logs here from start till `kill -INT`
[libusb.trace.gz](https://github.com/libuvc/libuvc/files/5847120/libusb.trace.gz)

Here is my setup. All packages were installed from apt repo.

Ubuntu Mate 20.04
```
 $ apt list --installed | grep uvc
guvcview/focal,now 2.0.6+debian-1build1 amd64 [installed]
libguvcview-2.0-2/focal,now 2.0.6+debian-1build1 amd64 [installed,automatic]
libuvc-dev/focal,now 0.0.6-1 amd64 [installed]
libuvc0/focal,now 0.0.6-1 amd64 [installed,automatic]
uvcdynctrl-data/focal,focal,now 0.2.4-1.1ubuntu2 all [installed,automatic]
uvcdynctrl/focal,now 0.2.4-1.1ubuntu2 amd64 [installed,automatic]

 $ apt list --installed | grep usb

ippusbxd/focal,now 1.34-2ubuntu1 amd64 [installed,automatic]
libgusb2/focal,now 0.3.4-0.1 amd64 [installed,automatic]
libusb-1.0-0/focal,now 2:1.0.23-2build1 amd64 [installed,automatic]
libusbmuxd6/focal,now 2.0.1-2 amd64 [installed,automatic]
libusbredirparser1/focal,now 0.8.0-1 amd64 [installed,automatic]
usb.ids/focal,focal,now 2020.03.19-1 all [installed,automatic]
usbmuxd/focal,now 1.1.1~git20191130.9af2b12-1 amd64 [installed]
usbutils/focal,now 1:012-2 amd64 [installed]

```
Here is some messages from `dmsg`

```
[121224.068005] usb 3-2.3.4.2: usbfs: process 210640 (camDa) did not claim interface 0 before use
[121367.695073] usb 3-2.3.4.2: usbfs: process 211020 (camDa) did not claim interface 0 before use
[121383.348247] uvcvideo: Found UVC 1.00 device PureThermal (fw:v1.3.0) (1e4e:0100)
[121383.349297] input: PureThermal (fw:v1.3.0): PureTh as /devices/pci0000:00/0000:00:1c.4/0000:04:00.0/usb3/3-2/3-2.3/3-2.3.4/3-2.3.4.2/3-2.3.4.2:1.0/input/input52

```
Perhaps this old issue might be something of [interest?](https://libusb-devel.narkive.com/1yGYVMgg/libusb-126-hanging-on-libusb-exit)

mrspirytus, 2021-01-21:
Here is a list of API I am using and the order they are called.

```
uvc_init(&_ctx, nullptr);
uvc_find_device(_ctx, &_dev, idVendor, idProduct, nullptr);
uvc_open(_dev, &_devHandle);
uvc_get_stream_ctrl_format_size(_devHandle, &_ctrl,
                                          UVC_FRAME_FORMAT_GRAY16,
                                          width, height, fps);
uvc_start_streaming(_devHandle, &_ctrl, frame_callback, this, 0);

// Process video callbacks

// Request shutdown
uvc_stop_streaming(_devHandle);
uvc_close(_devHandle);
uvc_unref_device(_dev);
uvc_exit(_ctx);                    <- It hangs here
```

rjgebis, 2021-01-26:
Now I noticed that sometimes it hangs on` uvc_stop_streaming(_devHandle);` 
I did setup 18.04 Ubuntu and it hangs most of the times on stop streaming. I see that others had such issue with uvc_stop_streaming() and condition variable. Was this ever properly addressed?

closed 2021-02-07 by mrspirytus