libuvc/libuvc #168: uvc_find_device: No such device (-4)

jeanlouis-villecroze, 2021-02-01:
I have compiled libuvc (after installing libusb via brew) on macOS 11.1, and the example fails to find the builtin iSight camera (Product ID: 0x8514, Vendor ID: 0x05ac):

uvc_find_device: No such device (-4)

Similarly, uvc_get_device() does not find any device. 

I have previously used libuvc successfully on older macOS (several years old). Is the library not supported on the release of macOS?

jeanlouis-villecroze, 2021-02-01:
So, this is probably and issue with libusb as libusb_get_device_list() returns 0. [This](https://github.com/libusb/libusb/issues/463#issuecomment-420134260) seems indicates that USB Entitlement is needed on that macOS version.

tpham3783, 2021-12-09:
Interesting enough, this is happening to me on Linux, Ubuntu 20.04 LTSC, kernel 5.4.0-71 and libusb-2:0.1.12-32

uvc_get_device() does not find any device.

please share if there is a work-around.  thanks

tpham3783, 2021-12-09:
NM,

I found my mistake:

>  libuv only binds with usbClass 14# 14, and subclass# 2. - device that I am trying to communicate is a FLIR-one pro, interface class# 255, and subclass 240. (aka proprietary)

I believe that I have to flash the firmware with the opensource version in order to utilize libuv.

sorry for the noise.

openastroproject, 2021-12-09:
Regarding the FLIR camera, if it's genuinely UVC but presents proprietary interface class and subclass values, there's already a small fudge around line 750 in src/device.c that handles a similar problem for some Imaging Source cameras that might be extended for yours?

closed 2026-09-06 by mcuee