libuvc/libuvc #169: How to connect to multiple cameras of same type? uvc_find_device()

mrspirytus, 2021-02-07:
I have two MS cameras connected to the system

```
Bus 001 Device 005: ID 045e:075d Microsoft Corp. LifeCam Cinema
Bus 001 Device 006: ID 045e:075d Microsoft Corp. LifeCam Cinema
```
Both have iSerial 0
```
libusb -v -d 045e:075d
```
**uvc_find_device()** does take serial but both cameras iSerial are 0

I am experimenting with stereo vision, and that is why I need to connect to both cameras.

jwlodek, 2021-02-07:
In my experience the only way to get around this is to run them in a singular sandbox such as in individual docker containers, and only mount one of them in each container.

mcuee, 2021-08-06:
Probably you can enhance uvc_find_device() to take in two more parameters: bus number and port number.

Ref: https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html

nuffer, 2022-02-16:
Same issue here, I have rewrite the `uvc_find_device` [function here](https://github.com/libuvc/libuvc/blob/37f00abd32b083a3d748f2a25c1c794578b51c16/src/device.c#L128) in order to iterate over the number of same camera connected. 

```c
uvc_error_t mys_uvc_find_hx_device( uvc_context_t *ctx, uvc_device_t **dev_hx, int vid, int pid, int idx) {

  uvc_error_t ret = UVC_SUCCESS;
  uvc_device_t **list;
  uvc_device_t *test_dev;
  int dev_idx;
  int found_dev;
  int found_idx;

  // Get list of devices
  ret = uvc_get_device_list(ctx, &list);
  if (ret != UVC_SUCCESS) {
    return ret;
  }

  dev_idx = 0;
  found_dev = 0;
  found_idx = 0;

  // iterate over the list
  while (!found_dev && (test_dev = list[dev_idx++]) != NULL) {
    uvc_device_descriptor_t *desc;

    // test capability to get descriptor
    if (uvc_get_device_descriptor(test_dev, &desc) != UVC_SUCCESS)
      continue;

    // test if corresponding vendor and product
    if ((!vid || desc->idVendor == vid) && (!pid || desc->idProduct == pid)) {
        if(found_idx == idx){
            found_dev = 1;
        }
        found_idx++;
    }

    uvc_free_device_descriptor(desc);
  }

  if (found_dev)
    uvc_ref_device(test_dev);

  uvc_free_device_list(list, 1);

  if (found_dev) {
    *dev_hx = test_dev;
    puts("[Success] to found corresponding hx camera");
    return UVC_SUCCESS;
  } else {
    puts("[Fail] to found corresponding hx camera");
    return UVC_ERROR_NO_DEVICE;
  }
}
```