libuvc/libuvc #171: How do I use ctrl-gen.py to generate control functions?

andreok, 2021-02-20:
Hello,

I would like to know how I could use ctrl-gen.py to generate a custom ctrl-gen.c for my camera.
What's the structure necessary for the input definition file?

Regards,
Andre

openastroproject, 2021-02-21:
I'd guess that standard-units.yaml might be the original input file?

andreok, 2021-02-21:
Thank you very much for pointing me to the standard-units.yaml file @openastroproject. 

I was checking again the standard-units.yaml and ctrl-gen.c and I noticed that none of them have the specific camera registry  values.
So, looks like I need to override the uvc_ct_ctrl_selector and uvc_pu_ctrl_selector enums with my specific control registry values?
But non-standard/extension control properties should be included into standard-units.yaml and mapped to a specific enum into libuvc.c?

If not asking too much, how do you handle this type of functionality in oaCapture for the Mac OS X platform?

Regards,
Andre

dmikushin, 2023-10-01:
```bash
python ./ctrl-gen.py -i ../standard-units.yaml def >ctrl-gen.c
```

closed 2026-09-06 by mcuee