libuvc/libuvc #173: Microsoft® LifeCam Studio issues with libuvc

mrspirytus, 2021-03-11:
I use this camera with Slack making calls. Somes, it works but then I stop working I have to reboot the system. The unplugging camera does not help. I did try switching from USB 3 to order USBbut it does not help
I am reporting this since the below message states to report it :)
Any know fixes? I am on Ubuntu Mate 20.04

Cheers

```
[21767.895006] uvcvideo: Found UVC 1.00 device Microsoft® LifeCam Studio(TM) (045e:0772)
[21767.895007] uvcvideo: Forcing device quirks to 0x80 by module parameter for testing purpose.
[21767.895007] uvcvideo: Please report required quirks to the linux-uvc-devel mailing list.
[21767.899003] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[21767.899548] uvcvideo: Failed to query (129) UVC probe control : -71 (exp. 26).
[21767.899549] uvcvideo: Failed to initialize the device (-5).
[21767.919902] usb 3-2.3.3.4: 3:1: cannot set freq 96000 to ep 0x82
[21767.935676] usb 3-2.3.3.4: 5:2: cannot get min/max values for control 2 (id 5)
[21767.939516] usbhid 3-2.3.3.4:1.4: can't add hid device: -71
[21767.939521] usbhid: probe of 3-2.3.3.4:1.4 failed with error -71
[21767.997918] usb 3-2.3.3.4: 3:1: usb_set_interface failed (-71)
[21768.003524] usb 3-2.3.3.4: 3:1: usb_set_interface failed (-71)
[21768.009116] usb 3-2.3.3.4: 3:1: usb_set_interface failed (-71)
[21768.014718] usb 3-2.3.3.4: 3:1: usb_set_interface failed (-71)
[21768.079349] usb 3-2.3.3.4: 3:1: usb_set_interface failed (-71)
[21768.575716] usb 3-2.3.3.4: 3:1: cannot set freq 48000 to ep 0x82
[21768.933434] usb 3-2.3.3.4: 3:1: cannot set freq 48000 to ep 0x82
[21770.018708] retire_capture_urb: 3207 callbacks suppressed
[21770.019704] xhci_hcd 0000:06:00.0: WARN: buffer overrun event for slot 14 ep 4 on endpoint
[21770.482700] xhci_hcd 0000:06:00.0: WARN: buffer overrun event for slot 14 ep 4 on endpoint
[21770.771697] xhci_hcd 0000:06:00.0: WARN: buffer overrun event for slot 14 ep 4 on endpoint
[21770.822668] xhci_hcd 0000:06:00.0: WARN: buffer overrun event for slot 14 ep 4 on endpoint
[21770.908692] xhci_hcd 0000:06:00.0: WARN: buffer overrun event for slot 14 ep 4 on endpoint
[21771.063663] xhci_hcd 0000:06:00.0: WARN: buffer overrun event for slot 14 ep 4 on endpoint
[21771.077685] xhci_hcd 0000:06:00.0: WARN: buffer overrun event for slot 14 ep 4 on endpoint
[21771.129662] xhci_hcd 0000:06:00.0: WARN: buffer overrun event for slot 14 ep 4 on endpoint
[21771.146687] xhci_hcd 0000:06:00.0: WARN: buffer overrun event for slot 14 ep 4 on endpoint

````

mcuee, 2021-08-17:
uvcvideo: Please report required quirks to the linux-uvc-devel mailing list. Not here.

mrspirytus, 2021-08-17:
Will do. Thank you

closed 2026-09-06 by mcuee